FrSkyRC/Firmware-Test #72: ACCESS 2.1 FPORT telemetry data bytestuffing bug

FieserKiller, 2020-06-28:
I have the following hardware and Software:
Taranis X9D+ 2019 SE
OpenTX 2.3.9
ISRM v2.1.0
FrSky Archer M+ running FPORT
Betaflight 4.2
BF Lua Script 1.5.0

Sending data frames from transmitter to betaflight vial FPORT telemetry fails CRC for specific values. Especially sending the byte sequences 125 and 126 fails every time.
It looks like this is a variation of an old r-xsr bug which was aknowledged and fixed by FrSky in 2018: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151 

It looks this time only FPORT protocol of r-xsr ACCESS firmware and Archer M+ firmware in FPORT mode is affected. Check the latest updates on this bug for some user reports and Betaflight developer analysis of the problem: 
https://github.com/betaflight/betaflight-tx-lua-scripts/issues/325

QuadStryke, 2020-07-04:
Could it be that you don't get RSSI aswell? I just opened a new issue. I set up FPort correctly, but I get no RSSI signal at all
https://github.com/FrSkyRC/Firmware-Test/issues/79

FieserKiller, 2020-07-09:
RSSI works fine.

QuadStryke, 2020-07-10:
How did you get it going? I used FPort and I get inputs and all that, but even though telemetry is enabled I dont get rssi. Only if I make a input and mix on my taranis for rssi. But that defeats the adavantage of Fport, because rssi should work straight away

FieserKiller, 2020-07-10:
No hard feelings but this is not some chit-chat disussion site but a github issue tracker. This is a place for focussed on-topic exchange and your rssi problems are a completely different topic. I won't answer to you and I'm asking you not to derail this issue anymore.

QuadStryke, 2020-07-10:
Well, then Frsky better fcking fix their firmware. Some one else has the same problem. And their XM+ ACCST 2.1.0 is fucked up to. There's an issue which is full of complaints

FieserKiller, 2020-10-31:
@Adelaxie I updated everything to the latest versions today:
- Betaflight to 4.2.4 (FF_RACEPIT_MINI target)
- ARCHER M+ to v2.1.5
- OpenTX to 2.3.10

This versions stayed the same because there is no update available:
- Taranis X9D Plus SE 2019  ISRM  v2.1.0
- Betaflight LUA Script 1.5.0

But the problem still persists. 
Everything works incl telemetry and lua script, the only thing that does not work is switching to Raceband channel 2 (5695MHz) and channel 3 (5732MHz). 
Betaflight devs still state the problem is clearly on FrSkys site (see analyse in my initial post). 
This ticket is open for 4 months now, do you have an update ehther you acknowledge the problem or not and whether its on the todo list to being fixed?

Adelaxie, 2020-11-03:
@FieserKiller 
This bug fix was about module & radio.
Will provide fixes together with the next release version.