FrSkyRC/Firmware-Test #73: accst 2.1.0 frsky x lite bind but does not connect to X4R

P-I-Engineer, 2020-06-28:
ISRM updated to 2.1

X4R updated to 2.1

rx will bind. After power cycle i get a flashing red light.

tested with multi protocol frsky X2 and it binds and connects. 

anyone else have this issue?

MIKE-DAILEY, 2020-06-28:
Works fine for me with FCC.

Mike

P-I-Engineer, 2020-06-29:
tested more:
X9 lite with isrm 2.1 firmware.

x4r accst 2.1: bind, no connect
xsr accst 2.1: bind, no connect

S6r accst 2.1: bind and connects, fully functional
r-xsr accst 2.1: bind and connects, still need to check for servo movements.

MIKE-DAILEY, 2020-06-29:
You might try to download the X9 Lite ISRM firmware again and re-flash the ISRM again.  I'm not having any issues.  

Mike

P-I-Engineer, 2020-06-29:
yeah, it's weird, i've flashed a couple times.

of the rx's i have, it's only the x4r and xsr having the issue.

you're on a x9 lite? right?

MIKE-DAILEY, 2020-06-29:
X9 Lite and X4R on FCC.  

Mike

P-I-Engineer, 2020-06-29:
ok, thanks for data point.

MIKE-DAILEY, 2020-06-29:
Are you using FCC?

Mike

P-I-Engineer, 2020-06-29:
yes, all fcc

P-I-Engineer, 2020-06-29:
weird, loaded lbt firmware on the x4r just as a test. nothing happened. Loaded fcc back on and it connected.

so i'm guess i'm good.

MIKE-DAILEY, 2020-06-29:
Great news!

Mike

P-I-Engineer, 2020-06-29:
spoke too soon, XSR wont' connect still

P-I-Engineer, 2020-06-30:
tried binding my x4r to a different model in my radio same issue. Bind but no connection.

P-I-Engineer, 2020-07-03:
https://www.youtube.com/watch?v=FzTpagYUBQU

P-I-Engineer, 2020-07-03:
i got some where. 

I got my x4r and xsr to bind, but, no matter what rx number i bind to, number 00 will connect and work.
so if i bind with rx number 13 assigned, it will flash red when after i power cycle, but if i change the number to 00 without
rebinding it connects and i get a green LED.

this is true for x4r and xsr

I tried different rx numbers on my other accst 2.1 rx's.
s6r and r-xsr will bind and connect on the rx number used during binding, so they're working correctly.

MIKE-DAILEY, 2020-07-03:
How does the X9 Lite work using an ACCESS receiver?

P-I-Engineer, 2020-07-03:
It works as expected, no issues that I can tell.

P-I-Engineer, 2020-07-04:
closing and making new issue

P-I-Engineer, 2020-07-25:
did you guys test with a regular x9 lite or x9 lite s? I know tested with an S and the rx's work correctly, but the cheaper x9 lite firmware is not doing model match correctly

MIKE-DAILEY, 2020-07-25:
X9 Lite and X9 Lite S both work OK for me. 

Mike

P-I-Engineer, 2020-07-25:
Thanks for the data point, I wanted to confirm variables.