FrSkyRC/Firmware-Test #75: ACCST S6R v2.1.0 Solid Green But Not Binding To X9D+ on OpenTX 2.3.9

tylerzudans, 2020-06-30:
S6R Firmware: v2.1.0, v2.1.0beta (both FCC)
X9D+ OpenTX Version: 2.3.9

Symptom: I start the bind from the X9D+ then power the receiver while holding the F/S button, and for a split second there's a very brief flash of red but it's solid green otherwise for the remaining time. This would imply that it's connected according to [this](http://rcdiy.ca/A/wp-content/uploads/2017/12/Slide06-768x432.png) chart from [this](http://rcdiy.ca/sxr-setup-s6r-s8r/) article, but it isn't really bound. I am unable to control any servo's or motors I have connected, unable to do even the first step in calibrating via the lua script, and receive no telemetry (the script also shows 0 for x, y, and z values). Whenever I power the RX in bind mode it repeats the same behavior mentioned above even when the TX is not in bind mode or off. When powered without any buttons pressed, it goes into signal lost mode with slow red flashes.
 
Cause: This all happened immediately after I successfully upgraded the firmware on the RX to v2.1.0. (FCC, and not the vtail version). Before I was binding successfully and getting telemetry, but the calibration script failed on the 4th step. Since I hadn't updated the firmware since getting the S6R I hoped that was the issue, but now I can't bind at all and don't know which RX firmware was on there originally.

Troubleshooting: 

1. I've tried multiple 2.1.0. versions including the 6/02 Beta with the same result.
2. I've tried using old RX firmware (S6R.frk from May 31st, 2019 with TX on 2.3.9) and the RX enters bind mode, but won't connect to the TX even after about a minute of waiting
3. I've tried using old TX firmware (OpenTX 2.2.2 from 2018 on X9D+ with RX on v2.1.0) and the RX enters bind mode, but won't connect to the TX even after about a minute of waiting
4. Made sure everything was fully charged, played around with channel numbers, and tried everything I could find like this problem on the OpenRC and RCGroups Forum but to no avail

Profile: In all cases, including when it worked Internal RF was set to mode XJT D16 with channel range CH1-16. 

Any insights or help troubleshooting is appreciated!

mpjf01, 2020-07-01:
If I understand you correctly the receiver connected to your transmitter correctly and operated normally except for an inability to calibrate to completion. This was Version 1 RF firmware. After you updated the firmware on the receiver to the production version 2.1.0 it will no longer bind.

The obvious potential culprit is a mismatch between transmitter and receiver RF firmware. You don't mention ever updating the transmitter RF firmware. You need to do this (update to 2.1.0) as well as all receivers that you use with it.

If you have updated the RF firmware on the transmitter to 2.1.0 you need to make sure that you used the same version (EU/non-EU) as is on the receiver. If you are unsure about what is on there update both the transmitter and receiver to 2.1.0 again using Non-EU or EU for both.

If you fix that and are still left with the calibration problem, repeat the procedure following the steps in the manual. The PC STK calibration (if you are using that) doesn't make it clear that it's finished, you can stop when it gets back to the first picture.

It won't help you much to be told this but I have installed 2.1.0 on my S6R and it binds and works normally.

tylerzudans, 2020-07-01:
That would make sense. I'm new to all this, so I misunderstood how the firmware on the transmitter works. I had updated OpenTX and the boot loader to the most recent versions, but didn't realize there was a seperate firmware for the internal RF module that needs to be flashed. I will try this and confirm if that's the issue.

tylerzudans, 2020-07-01:
That worked, thanks for the help!

closed 2020-07-01 by tylerzudans

mpjf01, 2020-07-02:
Glad you got it sorted out.