FrSkyRC/Firmware-Test #79: New Archer M+ no RSSI FPort

QuadStryke, 2020-07-04:
Hello,

I'm am so mad, that I have to write her. First, I got me a Xm+. And guess what, Frsky fcked up the firmware. There's an issue on github where people have 80 and lower rssi while standing next to the quad and extremely low rssi when flying away. Anyways, Frsky did not manage to fix it. So I though, well I buy the new AM+ Access (my radio is Qx7s Access) and there won't be any problems. But again, Frsky fcked it up.
My FC is a Crazybee F4R V3. I soldered a cable from TX1 to the inverted S.port. I get stick inputs and everything is fine. Except Rssi.
I chose FPort in Betaflight, enabled telemetry etc. Normally you should get rssi instantly. But guess what - no. I don't get any rssi signal, it's always 0. If I program an RSSI Input and Mixer in Otx, it works. But with FPort you are supposed to disbale rssi aux.

So, I bought another Frsky product because the one before was shitty and now the replacement is shitty aswell. 

I hope you fix that ASAP!

A mad customer

P-I-Engineer, 2020-07-04:
i think it's all the access f.port at this point.
rx6r doesn't work either

the only one i've got to work correctly is r-xsr beta firmware from a different issue in here.

Adelaxie, 2020-11-03:
Please check the fix from the latest firmware download from the website

closed 2020-11-03 by Adelaxie