FrSkyRC/Firmware-Test #81: S8R with MLVSS - wrong voltage values

haerel, 2020-07-10:
Horus X12S
OpenTX 2.3.9
S8R_ACCST_2.1.0_LBT.frk 2020-06-24
MLVSS Sensor

Log file shows many lower voltage peaks. Three different S8R and three different sensor modules, all have the same behavior. 
Cel_ is the LiPo cell with the lowest voltage.

[SuperGuppy-2020-07-09.zip](https://github.com/FrSkyRC/Firmware-Test/files/4901396/SuperGuppy-2020-07-09.zip)

ululi1970, 2020-07-17:
Have you tried with a different battery? This would rule out potential issues with the balancing port wires...

haerel, 2020-07-17:
This problem occurs with all my models with S8R and MLVSS as well as with all my LiPos (8x 6S5000, 6x 4S2200). Models with G-RX8 receiver and MLVSS work fine.

RetoD7, 2020-08-27:
@haerel 
Have you still your original iXJT module in your X12S? Or have you changed to the new ISRM (ACCESS upgrade) module?