FrSkyRC/Firmware-Test #83: wrong telemetry data with ACCST + ACCESS

RetoD7, 2020-07-17:
Rarely there are still some wrong telemetry data with different RX and with ACCST and ACCESS.
With an openXsensor I see more suche errors, then with other sensors as the MLVSS or SM GPS-Logger 3. This could be because of high data volume/rate or could also be an additional external problem.

During tests with openXsensor and Arduino-S.Port-Lib. I noticed some strange behavior when I forgot to stop Sensor Search in OpenTX. From time to time there are some new non existent sensors discovered and directly get lost. So this are some single wrong sensor data packets. Mostly there are just some bit errors to existing sensors. (other sensor ID/type or device ID ) So I made some more tests to try to locate the problem. 

I suspect some telemetry (S.Port) data handling in the RX. But it could be everything
This problem could be related to "[ACCESS 2.1 Telemetry Freeze mid flight #34](https://github.com/FrSkyRC/Firmware-Test/issues/34)"

_Because my openXsensors and arduino-projects could also cause some problems, the following cases are without them._ 

**TX: X10S Express, ISRM 2.1.0EU (LBT), OpenTX 2.3.9**

**case 1:**
**RX: G-RX6 FW-ACCESS_G-RX6_v2.1.0_200519
Sensor: MLVSS**

Wrong Cels, c1 and c2 Data. Cels Error follows about 300ms later than c1, c2 Erros, this seems strange to me. I hope it helps to locate the error.

`Date,Time,RSSI(dB),RxBt(V),A2(V),VFR(%),Cels(V),c1(V),c2(V),c3(V),Alt(m),VSpd(m/s),Sei,Hド,Gas,Que,S1,6P,S2,EX1,EX2,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,TxBat(V)
2020-07-16,00:28:39.860,81,4.9,12.2,100,11.53,3.84,3.85,3.84,151.8,0.0,-2,-1,-1022,2,-85,-767,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:39.960,85,4.9,12.2,100,11.53,3.84,3.85,3.84,151.8,0.0,-2,-1,-1024,1,-86,-768,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.060,85,4.9,12.2,100,11.53,3.84,3.85,3.84,151.8,0.0,-2,-1,-1022,2,-86,-768,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.160,85,4.9,12.2,100,11.53,5.39,0.51,3.84,151.8,0.0,-2,-1,-1022,1,-85,-768,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.260,84,4.9,12.2,100,11.53,5.39,0.51,3.84,151.7,0.0,-2,-1,-1024,1,-85,-767,1024,0,0,-1,640,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.360,84,4.9,12.2,100,11.53,5.39,0.51,3.84,151.7,0.0,-2,-1,-1024,2,-86,-768,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.460,84,4.9,12.2,100,9.75,5.39,0.51,3.85,151.7,0.0,-2,-1,-1022,1,-86,-768,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.560,84,4.9,12.2,100,9.75,5.39,0.51,3.85,151.8,0.0,-2,-1,-1024,1,-85,-767,1024,0,0,-3,640,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.660,84,4.9,12.2,100,9.75,5.39,0.51,3.85,151.8,0.0,-2,-1,-1024,1,-85,-767,1024,0,0,-3,640,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.760,85,4.9,12.2,100,9.75,5.39,0.51,3.85,151.8,0.0,-2,-1,-1022,2,-85,-767,1024,0,0,-3,640,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.860,85,4.9,12.2,100,9.75,3.84,3.86,3.85,151.8,0.0,-2,-1,-1022,1,-85,-768,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:40.960,85,4.9,12.2,100,9.75,3.84,3.86,3.85,151.8,0.0,-2,-1,-1024,1,-85,-768,1024,0,0,-1,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:41.060,83,4.9,12.2,100,9.75,3.84,3.86,3.85,151.8,0.0,-2,-1,-1022,1,-85,-767,1024,0,0,-3,640,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:41.160,83,4.9,12.2,100,11.55,3.84,3.86,3.85,151.8,0.0,-2,-1,-1024,1,-85,-768,1024,0,0,-1,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:41.260,83,4.9,12.2,100,11.55,3.84,3.86,3.85,151.8,0.0,-2,-1,-1022,1,-86,-768,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:41.360,87,4.9,12.2,100,11.55,3.84,3.85,3.85,151.7,0.0,-2,-1,-1022,1,-86,-767,1024,0,0,-3,640,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:41.460,87,4.9,12.2,100,11.55,3.84,3.85,3.85,151.8,0.0,-2,-1,-1022,1,-86,-767,1024,0,0,-3,641,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
2020-07-16,00:28:41.560,87,4.9,12.2,100,11.55,3.84,3.85,3.85,151.8,0.0,-2,-1,-1022,1,-86,-767,1024,0,0,-3,640,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.2
`

**case 2:**
**RX: G-RX6 FW-ACCESS_G-RX6_v2.1.0_200519
Sensor: MLVSS**

Non exisiting sensor discoverd:
[image]

(This sensor "101A" I saw also in some other tests I think it's related to G-RX6. The MVLSS was not involved in every case, but I'm not sure. Some of the tests were with ACCST and not ACCESS.)

I will try to report more "well" documented cases.

haerel, 2020-07-17:
I have the same problem.
Case 2 "Non-existent sensor detected" also occurs on my models with S8R. A lot of sensors are added that do not exist.

TX: X12S, upgrade ISRM 2.1.0EU (LBT), OpenTX 2.3.9

RetoD7, 2020-07-24:
I have made some more tests and also checked some of my flight logs. So there are some combinations, where I don't see telemetry errors (while many hours).
**okay, no errors seen:**
- RX8R Pro ACCST 2.1.0 + SM GPS-Logger 3
- R-XSR ACCESS 2.1.0 + RB-10 + SM GPS-Logger 3
- RX6R ACCESS 2.1.1 + MLVSS + SM GPS-Logger 3
- G-RX6 ACCTS 2.1.2 + MLVSS 

But here is an other combination **with some telemetry errors:**
**G-RX6 ACCTS 2.1.2 + MLVSS + SM GPS-Logger 3**
(inhouse test, so no GPS signal)

non existing sensors:
[image]

wrong data, some single "spikes":
**VSpd (from SM GPS Logger3): 4362076.1**
**AccZ(g): 167813.38 (could be a black hole ...)**
**Dir:  270139392 (but only can be in the range 0-360)**
**GSpd(kmh): 402.6 (there was no GPS-Signal  (no SAT) so this was also a telemetry error)**
```
Date,Time,RSSI(dB),RxBt(V),A2(V),VFR(%),Cels(V),c1(V),c2(V),c3(V),GPS,Dir,DirR,VSpd(m/s),AccX(g),AccY(g),AccZ(g),Alt(m),GAlt(m),GSpd(kmh),Hdg(@),Date,Dist,Sat,Diff,Alt(m),VSpd(m/s),Sei,Hド,Gas,Que,S1,6P,S2,EX1,EX2,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,TxBat(V)
2020-07-24,13:07:00.040,87,5.3,0.0,0,11.64,3.88,3.89,3.87,,0,0,0.0,-0.03,0.03,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,1.2,0.1,-1,1,-1024,-2,-314,-767,0,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,8.0
2020-07-24,13:07:00.140,87,5.3,0.0,0,11.64,3.88,3.89,3.87,,0,0,4362076.1,-0.03,0.03,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,1.2,0.1,-1,1,-1024,-1,-314,-767,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,8.0
2020-07-24,13:07:00.240,87,5.3,0.0,0,11.64,3.88,3.89,3.87,,0,0,4362076.1,-0.03,0.03,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,1.1,0.1,-1,1,-1024,-1,-314,-767,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,8.0
2020-07-24,13:07:00.340,86,5.3,0.0,0,11.64,3.88,3.89,3.87,,0,0,0.0,-0.03,0.03,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,1.2,0.1,-1,1,-1024,-1,-314,-767,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,8.0
......
2020-07-24,13:29:59.380,90,5.2,0.0,0,11.63,3.89,3.88,3.86,,0,0,0.0,-0.03,0.03,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.4,0.2,-1,1,-1024,-1,-87,-767,0,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:29:59.480,93,5.2,0.0,0,11.63,3.89,3.88,3.86,,0,0,0.0,-0.03,0.03,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.5,0.2,-1,1,-1024,-2,-87,-767,0,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:29:59.580,93,5.2,0.0,0,11.63,3.89,3.88,3.86,,0,0,0.0,-0.03,0.03,167813.38,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.4,0.1,-1,1,-1024,-1,-87,-767,-1,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:29:59.680,93,5.2,0.0,0,11.63,3.89,3.88,3.86,,0,0,0.0,-0.03,0.03,167813.38,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.4,0.1,-1,1,-1024,-1,-87,-767,-1,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:29:59.780,92,5.2,0.0,0,11.63,3.89,3.88,3.86,,0,0,0.0,-0.03,0.03,167813.38,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.4,0.1,-1,1,-1024,-1,-87,-768,-1,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:29:59.880,92,5.2,0.0,0,11.63,3.89,3.88,3.86,,0,0,0.0,-0.03,0.03,167813.38,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.5,0.0,-1,1,-1024,-1,-87,-768,0,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:29:59.980,92,5.2,0.0,0,11.63,3.88,3.89,3.86,,0,0,0.0,-0.03,0.03,167813.38,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.4,0.0,-1,1,-1024,-1,-87,-767,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:30:00.080,89,5.2,0.0,0,11.63,3.88,3.89,3.86,,0,0,0.0,-0.03,0.03,167813.38,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.4,0.0,-1,1,-1024,-2,-87,-767,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:30:00.180,89,5.2,0.0,0,11.63,3.88,3.89,3.86,,0,0,0.0,-0.03,0.03,167813.38,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.4,0.0,-1,1,-1024,-1,-87,-768,-1,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:30:00.280,89,5.2,0.0,0,11.63,3.88,3.89,3.86,,0,0,0.0,-0.03,0.03,167813.38,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.3,0.0,-1,1,-1024,-1,-87,-768,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:30:00.380,94,5.2,0.0,0,11.63,3.88,3.89,3.86,,0,0,0.0,-0.03,0.03,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.3,-0.1,-1,1,-1024,-1,-87,-768,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-07-24,13:30:00.480,94,5.2,0.0,0,11.63,3.88,3.89,3.86,,0,0,0.0,-0.03,0.03,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.3,-0.1,-1,1,-1024,-1,-87,-768,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
......
2020-07-24,14:08:25.240,91,5.1,0.0,0,11.66,3.89,3.89,3.88,,0,0,0.0,-0.03,0.03,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,9.4,0.0,-1,0,-1024,0,-87,-768,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
2020-07-24,14:08:25.340,91,5.1,0.0,0,11.66,3.89,3.89,3.88,,0,0,0.0,-0.03,0.03,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,9.4,0.0,-1,1,-1024,0,-87,-768,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
2020-07-24,14:08:25.440,92,5.1,0.0,0,11.66,3.89,3.89,3.88,,270139392,0,0.0,-0.03,0.03,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,9.4,0.0,-1,1,-1024,-1,-87,-768,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
2020-07-24,14:08:25.540,92,5.1,0.0,0,11.66,3.89,3.89,3.88,,270139392,0,0.0,-0.03,0.03,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,9.4,0.0,-1,0,-1024,0,-87,-768,-1,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
[... cut ...]
- more errors with additional RX functions (vario of G-RX8) (or gyro etc. if I see issue "S8R with MLVSS - wrong voltage values #81")
(With openXsensor I see the same kind of errors in combination with all RX. This could be an error of my openXsensor itself, or caused by a higher data volume or so ...)

I have not seen wrong data from internal sensors of G-RXx (RSSI, RxBt, A2, Vspd), so the problem could be the S.Port handling. I assume some performance / buffer problem or timing collision (interrupt handling?).

RetoD7, 2020-07-24:
I just repeated the above test where 8 telemetry errors occured during about 3 hours. But this time I switched off the internal variometer function of the G-RX6. A totally different result:

**G-RX6 ACCTS 2.1.2 (internal variometer off) + MLVSS + SM GPS-Logger 3**: no telemetry error during 4 hours

I think the main difference is not that there are 2 telemetry parameters less. I think the essential difference is the omitted variometer function and so a lower chance of some timing or performance conflicts.

RetoD7, 2020-08-14:
Yesterday my new Archer GR8 arrived and after a "short" test I found out it's **also affected:  Archer GR8 (ACCESS 2.1.0) + MLVSS + SM GPS-Logger 3**

Some other **combinations where I saw wrong telemetry data**:
G-RX6 FW-ACCESS_G-RX6_v2.1.0_200519 (vario on) + MLVSS + SM GPS-Logger 3
G-RX6 FW-ACCESS_G-RX6_v2.1.0_200519 (vario on) + SM GPS-Logger 3 + RB10
G-RX6 FW-ACCESS_G-RX6_v2.1.0_200519 (vario on) + MLVSS + RB10

In many (or nearly all?) cases I found the **Hex Code Sequenze "10 1A" or only "1A"**. This in the sensor ID (see screenshots above) or in the data encoding.
some Examples from yesterday test: G-RX6 FW-ACCESS_G-RX6_v2.1.0_200519 (vario on) + MLVSS + SM GPS-Logger 3
```
2020-08-13,13:14:03.930,100,5.0,12.2,100,11.63,3.88,3.88,3.87,,0,0,0.0,-0.04,0.01,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,-0.2,0.0,0,2,-1022,-2,387,-768,0,0,0,-1,1024,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.7
2020-08-13,13:14:04.030,100,5.0,12.2,100,11.63,3.88,3.88,3.87,,0,1055232,0.0,-0.04,0.01,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,-0.2,0.0,0,2,-1022,-1,387,-768,-1,0,0,-3,1024,1,0,1,-1,-1,-1,-1,-1,0x0000000000000000,7.7

```-> **1055232 = 0x101A00**

```
2020-08-13,13:36:05.220,69,4.9,12.2,100,11.63,3.88,3.88,3.87,,0,0,0.0,-0.03,0.01,1.03,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,-0.9,0.0,-3,-1,-1016,2,-116,-767,0,0,0,-3,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.7
2020-08-13,13:36:05.320,69,4.9,12.2,100,11.63,3.88,3.88,3.87,,0,0,0.0,-0.03,0.01,1.03,0.0,0.0,0.0,2701393.92,2048-00-00 00:00:00,0,0,0,-0.9,0.0,-4,-1,-1016,2,-116,-767,0,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.7

```-> **2701393.92 -> 270139392 = 0x101A0000**

```
2020-08-13,13:41:04.530,68,4.9,12.2,100,11.63,3.88,3.88,3.87,,0,0,0.0,-0.03,0.02,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,-0.8,0.0,-4,-1,-1016,2,-116,-767,0,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.7
2020-08-13,13:41:04.630,69,4.9,12.2,100,11.63,3.88,3.88,3.87,,0,0,0.0,-0.03,0.02,1.04,0.0,0.0,0.0,4362076.16,2048-00-00 00:00:00,0,0,0,-0.8,0.0,-4,-1,-1016,2,-116,-767,0,0,0,-1,1024,1,0,1,-1,-1,-1,0,-1,0x0000000000000000,7.7

```-> **4362076.16 -> 436207616 = 0x1A000000**

And also the wrong data in my earlier reports above:
**AccZ(g): 167813.38 = 0x0100101A 
Dir: 270139392 = 0x101A0000**

So this error exists in combination with different sensors and not only with on specific external sensor/device. But it seems more directly related to the internal sensors of the G-RX6 or GR8. If the count of external devices or count of sensors values maters or if this only increases to possibility of an error I don't know.

I'm very happy how V2.1 works at the moment, but I hope you will also correct the remaining little but annoying bugs.

RetoD7, 2020-08-17:
Also with disabled internal Vario of the G-RX6 there are some telemetry errors of the connected S.Port Sensors. But enabling or disabling makes a big difference in telemetry error rate. With 3 Sensors at the S.Port there are:
**- Vario off: 5 Errors in 6 hours (about 1 err/hr)
- Vario on: 22 Errors in 3 hours (about 7 err/hr)**

Details of this Test:
**TX: X10S Express, ISRM 2.1.0EU (LBT), OpenTX 2.3.9
RX: G-RX6 FW-ACCESS_G-RX6_v2.1.0_200519
Sensors: MLVSS + RB10 + SM GPS-Logger 3
Distance RX-TX: about 3m
Inhouse, incl. WLAN etc. arround**

### Telemetry Erros with disabled Vario:
-> 1 false sensor found: "1A00" ID: 25

```2020-08-16,00:04:31.150,84,4.8,12.1,100,11.65,3.89,3.88,3.88,,0,0,0.0,-0.04,0.04,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.93,0.00,0.02,0.00,0,64800,0,0,-2,1,-1020,0,-78,-767,0,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.4
2020-08-16,00:04:31.250,84,4.8,12.1,100,11.65,3.89,3.88,3.88,,0,0,0.0,-0.04,1845493.81,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.97,0.00,0.02,0.00,0,64800,0,0,-2,0,-1020,0,-78,-767,0,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.4```
-> AccY: 1845493.81 = 0x0B000005

```2020-08-16,00:11:15.060,86,4.8,12.1,100,11.63,3.88,3.89,3.86,,0,0,0.0,-0.04,0.04,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.90,0.00,0.02,0.00,0,64800,0,0,-2,0,-1018,0,-78,-767,0,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.4
2020-08-16,00:11:15.160,86,4.8,12.1,100,11.63,3.88,3.89,3.86,,0,0,0.0,-0.04,0.04,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.90,0.00,0.02,0.00,0,784,0,0,-2,0,-1020,0,-78,-767,0,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.4
2020-08-16,00:11:15.260,87,4.8,12.1,100,11.63,3.88,3.89,3.86,,0,0,0.0,-0.04,0.04,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.95,0.00,0.02,0.00,0,64800,0,0,-2,0,-1020,0,-78,-767,0,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.4```
-> RBS: 784 = 0x0310

```2020-08-16,22:31:33.180,82,4.8,12.1,100,11.61,3.88,3.88,3.86,,0,0,0.0,-0.03,0.04,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.94,0.00,0.02,0.00,0,64800,0,0,-3,0,-1018,1,105,1022,-2,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
2020-08-16,22:31:33.280,82,4.8,12.1,100,11.61,3.88,3.88,3.86,,0,0,0.0,-0.03,0.04,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.94,0.00,0.02,0.00,0,6688,0,0,-2,0,-1018,1,106,1022,-1,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
2020-08-16,22:31:33.380,82,4.8,12.1,100,11.61,3.88,3.88,3.86,,0,0,0.0,-0.03,0.04,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.99,0.00,0.02,0.00,0,64800,0,0,-2,0,-1018,1,106,1022,-2,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.8```
-> RBS: 6688 = 0x1A20

```2020-08-16,23:40:30.160,81,4.8,12.1,100,11.62,3.88,3.87,3.87,,0,0,0.0,-0.04,0.04,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.87,0.00,0.02,0.00,0,64800,0,0,-370081008,-2,0,-1020,1,105,1022,-1,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.7
2020-08-16,23:40:30.260,79,4.7,12.1,100,11.62,3.88,3.87,3.87,,0,0,0.0,-0.04,0.04,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,184553472,0,4.85,0.00,0.02,0.00,0,64800,0,0,-370081008,-2,0,-1020,0,105,1022,-1,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.7```
-> SAT: 184553472 = 0x0B001000

### Telemetry Erros with enabled Vario:
-> 7 false sensors found and lost, and updated 2 of them, so total 9 errors
![screen-2020-08-17-112229](https://user-images.githubusercontent.com/65849766/90423177-50113580-e0bc-11ea-91e9-e6a259471c8a.jpg)

```2020-08-17,09:42:58.870,82,4.6,12.1,100,11.63,3.88,3.88,3.87,,0,0,0.0,-0.03,0.05,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.69,0.00,0.02,0.00,0,64800,0,0,4.1,0.0,-1,1,-1021,-1,439,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5
2020-08-17,09:42:58.970,82,4.6,12.1,100,11.63,3.88,3.88,3.87,,0,436207616,0.0,-0.03,0.05,1.04,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.74,0.00,0.02,0.00,0,64800,0,0,4.2,0.0,-1,1,-1021,-1,439,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5```
-> DirR: 436207616 = 0x1A000000

```2020-08-17,09:57:29.890,82,4.6,12.1,100,11.62,3.87,3.88,3.86,,0,0,0.0,-0.03,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.69,0.00,0.01,0.00,0,64800,0,0,4.1,0.0,-1,1,-1021,0,440,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5
2020-08-17,09:57:29.990,82,4.6,12.1,100,11.62,3.87,3.88,3.86,,0,270139392,0.0,-0.04,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.65,0.00,0.02,0.00,0,64800,0,0,4.1,0.0,-1,0,-1021,-1,439,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5```
-> DirR: 270139392 = 0x101A0000

```2020-08-17,09:29:35.630,78,4.6,12.1,100,11.62,3.89,3.89,3.86,,0,0,0.0,-0.03,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.70,0.00,0.02,0.00,0,64800,0,0,2.6,-0.1,-1,1,-1021,0,439,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5
2020-08-17,09:29:35.730,82,4.6,12.1,100,11.62,3.89,3.89,3.86,,0,0,2702049.2,-0.03,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.70,0.00,0.02,0.00,0,64800,0,0,2.6,-0.1,-1,1,-1021,0,439,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5
2020-08-17,09:29:35.830,82,4.6,12.1,100,11.62,3.89,3.89,3.86,,0,0,2702049.2,-0.03,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.72,0.00,0.02,0.00,0,64800,0,0,2.7,0.0,-1,1,-1021,0,439,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5
[... cut ...] 1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5
2020-08-17,09:59:00.690,77,4.6,12.1,100,11.64,3.88,3.88,3.88,,0,0,0.0,-0.03,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,4.64,0.00,0.02,0.00,0,64800,0,0,4.3,0.0,-1,1,-1021,-1,440,1022,-2,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.5```
-> RB1C+RB2C: 6656,16: 0x1A000010

**Many of the erros contain 0x1A or a part of 0x0100101A but there are also other error data. But it shows the errors are not random, they are systematic in some kind.**

RetoD7, 2020-08-18:
The same Test with an **RXSR** in place of the G-RX6 shows a completely different picture of the telemetry error!? During 30 Minutes many "existing" sensors have been found with their correct value, but with the wrong device ID 2 (The OpenTX sensor list shows it as 1, in the sensor details its 2, as the MLVSS is/should be.)
[image]

Details of this Test:
TX: X10S Express, ISRM 2.1.0EU (LBT), OpenTX 2.3.9
RX: RXSR V2.1.0 (frsk-file time stamp 6.5.2020) **_please change the version-string if there are any changes!!!_**
Sensors: MLVSS + RB10 + SM GPS-Logger 3
Distance RX-TX: about 3m
Inhouse, incl. WLAN etc. arround

RetoD7, 2020-08-19:
### Test: Archer GR8 (ACCESS 2.1.0) + Sensors: MLVSS + RB10 + SM GPS-Logger 3
TX: X10S Express, ISRM 2.1.0EU (LBT), OpenTX 2.3.9
RX: RXSR V2.1.0 (frsk-file time stamp 6.5.2020) please change the version-string if there are any changes!!!
Distance RX-TX: about 3m
Inhouse, incl. WLAN etc. arround
Duration: 4 hours, 50 min.

**6 non existing sensors found and 2 value update of them
2 data errors**

[image]

```
2020-08-19,12:42:39.490,83,4.9,1.8,100,11.64,3.88,3.88,3.88,8.7,0.0,0.02,0.00,0,64800,0,0,4.82,0.00,0,,0,0,0.0,0,-0.02,0.00,0.0,0.0,0.0,0.00,2020-08-19 12:42:57,0,1.07,-3,-1,-1018,1,-34,1022,-1,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
2020-08-19,12:42:39.590,83,4.9,1.8,100,11.64,3.88,3.88,3.88,8.7,0.0,0.02,0.00,0,64800,0,0,4.82,0.00,0,,0,0,0.0,0,-0.02,0.00,0.0,0.0,0.0,0.00,2020-08-19 12:42:57,0,2701394.98,-3,-1,-1018,1,-34,1022,-1,0,0,-4,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
```
-> AccZ: 2701394.98 = 0x101A006A

```
2020-08-19,14:25:53.210,84,4.8,1.8,100,11.61,3.87,3.87,3.87,15.9,0.0,0.02,0.00,0,64800,0,0,4.71,0.00,0,,0,0,0.0,0,-0.02,-0.01,0.0,0.0,0.0,0.00,2020-08-19 14:26:11,0,1.07,-3,-1,-1018,1,-34,1022,-1,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.6
2020-08-19,14:25:53.310,84,4.8,1.8,100,11.61,3.87,3.87,3.87,15.9,0.0,0.02,0.00,0,64800,0,0,6.80,7.84,0,,0,0,0.0,0,-0.02,-0.01,0.0,0.0,0.0,0.00,2020-08-19 14:26:11,0,1.07,-3,-1,-1018,1,-34,1022,-1,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.6
2020-08-19,14:25:53.410,84,4.8,1.8,100,11.61,3.87,3.87,3.87,16.0,0.0,0.02,0.00,0,64800,0,0,6.80,7.84,0,,0,0,0.0,0,-0.02,-0.01,0.0,0.0,0.0,0.00,2020-08-19 14:26:11,0,1.07,-3,-1,-1017,2,-34,1022,-1,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.6
2020-08-19,14:25:53.510,83,4.8,1.8,100,11.61,3.87,3.87,3.87,16.0,0.0,0.02,0.00,0,64800,0,0,4.72,0.00,0,,0,0,0.0,0,-0.02,-0.01,0.0,0.0,0.0,0.00,2020-08-19 14:26:11,0,1.07,-3,-1,-1018,1,-34,1022,-1,0,0,-3,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.6
```
-> RB1V, RB1A: 6.80,7.84 -> 7.84 *100 * 2^16 + 6.80 *1000 -> 0x03101A90   (+1..9 "round/cut")

**The Archer GR8 behaves like the G-RX6. The errors er very similar, with might be a better performance.**

RetoD7, 2020-08-19:
### Test: RX6R ACCST 2.1.1 + Sensors: MLVSS + RB10 + SM GPS-Logger 3
TX: X10S Express, ISRM 2.1.0EU (LBT), OpenTX 2.3.9
Distance RX-TX: about 3m
Inhouse, incl. WLAN etc. arround
Duration: 4 hours, 45 min.

no non existing sensor found
1 data error:
```
2020-08-19,19:00:40.080,88,4.8,12.1,11.62,3.88,3.87,3.87,,0,0,0.0,-0.03,0.06,1.07,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,0,0,4.93,0.00,0.02,0.00,0,64800,-3,-1,-1018,1,-34,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-08-19,19:00:40.180,88,4.8,12.1,11.62,3.88,3.87,3.87,,0,0,0.0,-0.03,0.06,1.07,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,0,0,4.92,66.56,0.02,0.00,0,64800,-3,-1,-1018,1,-34,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-08-19,19:00:40.280,90,4.8,12.1,11.62,3.88,3.87,3.87,,0,0,0.0,-0.03,0.06,1.07,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,0,0,4.92,66.56,0.02,0.00,0,64800,-3,-1,-1018,1,-34,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
2020-08-19,19:00:40.380,90,4.8,12.1,11.62,3.88,3.87,3.87,,0,0,0.0,-0.03,0.06,1.07,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,0,0,4.92,0.00,0.02,0.00,0,64800,-3,-1,-1018,2,-34,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.9
```
-> RB1A: 66.56 *100 = 0x1A00

### Test: RX6R ACCESS 2.1.0 + Sensors: MLVSS + RB10 + SM GPS-Logger 3
TX: X10S Express, ISRM 2.1.0EU (LBT), OpenTX 2.3.9
Distance RX-TX: about 3m
Inhouse, incl. WLAN etc. arround
Duration: 6 hours

1 non existing sensor found: sensor ID: 101A
1 data error:
```
2020-08-18,09:08:12.950,77,4.9,12.1,100,11.63,3.87,3.88,3.87,,0,0,0.0,-0.03,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,0.02,0.00,0,64800,0,0,4.93,0.00,-1,0,-1021,-1,-34,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
2020-08-18,09:08:13.050,77,4.9,12.1,100,11.63,3.87,3.88,3.87,,0,0,0.0,-0.03,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,0.02,66.56,0,64800,0,0,4.93,0.00,-1,0,-1021,0,-34,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
2020-08-18,09:08:13.150,77,4.9,12.1,100,11.63,3.87,3.88,3.87,,0,0,0.0,-0.03,0.05,1.05,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,0,0,0.02,0.00,0,64800,0,0,4.97,0.00,-1,0,-1021,0,-34,1022,-2,0,0,-1,1024,1,1,1,-1,-1,-1,0,-1,0x0000000000000000,7.8
```
-> RB2A: 66.56 *100 = 0x1A00

### Conclusion:
Comparing to other RX the RX6R works well. **But I think it also makes clear that the telemetry error exists in all RX. It's just a mater of sensor count and time until a telemetry error occurs. But the good thing is, there is no error with random data, as far as I can estimate. So no RF problem. (But I have tested with strong signals only)**
I can not eliminate that the error is located on the TX side (ISRM, etc.), but I don't think so, because the different RX (and RX setup) behaves so different.

RetoD7, 2020-08-27:
The often seen code 0x0100101A means the S.Port sequenze 0x1A 0x10 0x00 0x01  (Little-Endian) and so is Alt Sensor from RX. In general 0x101A means RX internal sensor (A2, Alt Vspd, VFR). There are also many other wrong telemetry values, but in my tests this sequences are often for some reason.

**In the found error values there is no start byte 0x7E before the Phyisical ID (0x1A). And also the Phyisical ID is missing its Checksum (0x1A -> 0xBA). Starting with the Phyisical ID there are at a random position overwritten telemetry data. Where, under which conditions can this happen?**

It seems there are also concerned systems with the XJT module (not just ISRM): [Incorrect Telemetry Values G-RX8_ACCST_2.1.0_LBT20200519 #53](url)
Vspd: 2702046.8 *100 = 0x101AFF08   (or because of round/cut: +0..9)

RetoD7, 2020-08-27:
Or could it be simply the other way around? The telemetry data are **read out of the buffer before they are written completely** into the buffer? So the first bytes are new data and the second part of the message are old values from "last buffer cycle"?

RetoD7, 2020-08-28:
For me the theory that **telemetry data are read out of the buffer before they are written completely into the buffer** makes sens:

- Explains different forms of incomplete and wrong data.
- Internal sensor values of the RX are not (or rarely) affected because the ere write at one in a short time.
- Reading data from S.Port takes some time and so the chance is bigger, that in case of the telemetry error, data are read before the are completed. So then the second part of the message consists of old "random" data from the buffer.
- In my tests I saw often parts of the Alt sensor message (see above) as second part in a faulty sensor message. The value of this sensor was more or less the only one who has changed in a greater range. Changing slowly some meters with a resolution of 100 steps per meter gives many different values. The chance to hit the checksum is 1:255 or even bigger with some different seen combinations above. (Also with a perfect RF connection.)

**So I assume this error is located somewhere in the RX software. (ACCST V2.1, ACCES V2.1 including Archer)** (But it's not impossible that the error is located elsewhere.)

I assume this telemetry error is also responsible for a larger number of not arriving sensor data, because many of the wrong telemetry messages are stopped by the wrong checksum. This could be up to some messages per second.

Certainly I don't know how your software is constructed, so my conclusion might be wrong. But I hope it's a help to find out which exactly situation or what event is causing this annoying telemetry errors.

The G-RX6 (ACCST or ACCESS) would be a good candidate for tests.

RetoD7, 2020-09-19:
I tested with some debug/trace functions in OpenTX. In **OpenTX arrive no telemetry data with wrong checksum or out of sync data**. (As far as I have understand and tested.) So the wrong telemetry data are packet together in the TX module or RX.

(In some cases I don't know how the correct checksum appears. It's possible that wrong data are completed with correct checksum somewhere in the TX module or RX.)

RetoD7, 2020-10-10:
Also with the new firmware **Archer-X 2.1.2 the telemetry error still exists**.

The collision of an internal sensor value (A2, Alt, VSpd) with an external sensor value (S.Port) is the most common error. **For me the only possible place where this has some significance is in the RX, so the error must be located there.**

I captured an informative error:
GR8 Archer-X 2.1.2 + RB10 + MLVSS + SM GPS-Logger3
```
2020-10-02,00:38:00.910,82,4.9,1.76,100,11.21,3.73,3.75,3.73,493.57,0.19,0.02,0.00,0,64800,0,0,4.88,0.00,0,,0,0,0.0,0,-0.05,0.00,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,1.03,-2,1,-1024,1,-289,-767,0,0,0,-1,1024,1,0,0,-1,-1,-1,-1,-1,0x0000000000000100,7.6
2020-10-02,00:38:01.010,81,4.9,1.76,100,11.21,3.73,3.75,3.73,493.57,0.01,0.02,0.00,0,64800,0,0,4.88,0.00,0,,0,0,0.0,0,-0.05,0.00,0.0,0.0,0.0,0.00,2048-00-00 00:00:00,0,1.03,-1060503296,-2,1,-1024,1,-289,-767,0,0,0,-1,1024,1,0,0,-1,-1,-1,-1,-1,0x0000000000000100,7.6
```
new sensor found: physical ID: 5  , sensor ID : 101A, value: -1060503296 = 0xc0ca0100
-> Telemetrie-Datastream: **7E 05 10 1A 10 00 01 CA C0**
-> correctly it should be": **7E 05 10**  _(6 Byte of SM GPS-Logger)_ 7E **1A 10 00 01 CA C0 00 00**
So the Alt sensordata of GR8 is destroying a value from the SM GPS-Logger3. But I can't find this exactly value alt=493.54 (0xc0ca) some seconds before this error. The alt value is some cm lower or higher before and after, but not exactly the same. So it is not possible its a left over in data buffer.
**So the write processes for the telemetry buffer are not protected against each other.**

But I'm not able to reconstruct all errors, so I don't know if other conflicts exists.

Nerenye, 2020-10-16:
We did report these errors for months now. They're not giving a .... about.

RetoD7, 2020-11-01:
Also with the new firmware Archer-X 2.1.5 the telemetry error still exists.

May I ask others to report this bug to Frsky, as FrSky seems thinking I'm the only one with this errors.

Nerenye, 2020-11-01:
I did report it months ago but they didn´t even answer. 
 #53 
Neither seems the community to much concerned about this issue. 
Ive Abandoned my gr receivers and for now using the external vario since it doesn´t have this bug.

RetoD7, 2020-11-03:
@Nerenye Do you have the Taranis qx7 or Taranis qx7 Access with an ISRM TX-Module?

Nerenye, 2020-11-03:
> @Nerenye Do you have the Taranis qx7 or Taranis qx7 Access with an ISRM TX-Module?

As I did answer you in the linked issue already. Its the qx7 accst but reliing on your data both are affected.

Adelaxie, 2020-11-12:
> Also with the new firmware Archer-X 2.1.5 the telemetry error still exists.
> May I ask others to report this bug to Frsky, as FrSky seems thinking I'm the only one with this errors.

@RetoD7  Thanks for your detailed tests, please write to me for further tests. thanks a lot. libo.xie@frsky-rc.com

RetoD7, 2021-01-31:
Also with the new firmware Archer-X 2.1.6 the telemetry error still exists.

I sent some additional test findings and hints to FrSky. But obviously they have other priorities.

current summary of my test results:
If the receiver has other internal telemetry values then RSSI and RxBt, there occur telemetry errors when external S.Port sensors are connected. This means the internal sensors A2, Alt, Vspd or VFR are causing data collisions with data from external S.Port sensors. the problem was already the same with the old protocoll V1.

Depending on receiver and S.Port sensor combination the error has different character and occurrence.
- some short single wrong telemetry values (spikes, unrealistic values depending on range)
- (single) slow updating telemetry value, up to the message "sensor lost"
- non existing sensors are found, if sensor detection is running (helps when testing, not a problem in normal use)

In other words: **If you want use external S.Port sensors without telemetry errors you can not use any Archer or ACCESS receiver in general and some other receiver types.**

For me are working RX8R pro and R-XSR with ACCST 2.1.0

Nerenye, 2021-09-03:
@RetoD7  it´s also a problem with the integrated Telemetry sensors on grx 8 and grx 6 on ACCST. 

I know it´s actualy not a major issue but its extremly anoying and makes logs of long duration flight almost useless if youre not willing to invest like 2 Hours to correct the data.

Is there any plan from FrSky to fix this issue at all? @Adelaxie 
I mean like i do kinda understand youir prio is on Archer and Tandem now but if RetoD7´s data is correct your archer protokoll is also plagued by this issue.
There should be some interesst to resolve this. It´s not exactly good advertising in general to have faulty telemetry data on a mjority of your rxes

Best Regards Patrik