FrSkyRC/Firmware-Test #84: R9 ACCESS - First channel is flawed and not updating with each packet

Enurian, 2020-07-17:
Since I got no answer through your contact form, I´m writing it here again:
There is a major issue with the ACCESS firmware that makes the first channel not a good option as a control channel (for switches it´s fine): It is missing a lot of updates, many times there are 20-30ms between new values instead of the 6,7ms that are usually between them. Result is a much higher latency on this channel and in case of a quadcopter the flightcontroller will have trouble to smoothen the signal.

So here you can see the bug:
Throttle on the first channel (channel mapping TAER1234). The upper red line shows the raw signal. The second graph shows the unwanted stairs in the throttle signal because updates are too scarce to smoothen it.
[image]

After finding out that only the first channel is "corrupted", I switched the mapping to 2AER1T34, so an aux channel is in the front. Then throttle looks like this, this is how it should be:
[image]

Note that this appears ONLY with ACCESS firmware! When I switch back to ACCST, the issue is not present!

If you want to test it, here is my current config:
- Original X-Lite on OpenTX 2.3.9
- R9M Lite and R9M Lite Pro modules (I have both, both tested, same behavior) on ACCESS 1.3.0 FLEX
- 5 different R9 MM and R9 MX receivers on ACCESS 1.3.1/1.3.2 FLEX
- Receivers are bound on 868MHz, 8 channels, Telemetry on.
- To record the raw input, I´m using Betaflight 4.2.0 + Blackbox with cli settings
set rc_smoothing_debug_axis = THROTTLE
set debug_mode = RC_SMOOTHING

alexdibi, 2020-07-19:
same problem here:
X9 Lite OpenTX 2.3.9
R9m Lite ACCESS 1.3.0 FLEX
R9 MX 1.3.2 FLEX Fport
my config is AETR
[image]

Adelaxie, 2020-07-21:
Will test on it . thanks

alexdibi, 2020-07-23:
News?

Enurian, 2021-01-16:
It has been half a year now and there is no fix, not even a new firmware version.
Do you plan to do anything about it or should we just accept that the first channel cannot be used for latency critical signals?!

alexdibi, 2021-01-16:
Search for expresslrs, if you want a firmware that works decently for frsky stuff and if there are any bugs they are fixed...😉