FrSkyRC/Firmware-Test #86: With the 2.1, channel 8 no longer works!

Devil-67, 2020-07-19:
With the 2.1, channel 8 no longer works!
Channels 1-7 and 9 work normally, I have not checked the remaining channels.
Back to firmware V1 and channel 8 works normally again.

Please fix the problem

Devil-67, 2020-07-19:
Another problem with the V2.10 and the XSR!
Switch on the transmitter and then the receiver
When the power supply is applied, the servos move briefly to max position and then only back to the actual middle position.

This is very, very bad, there are some flaps which are then destroyed on their mechanical stop.
Or burn the servos when they hit the stop.

@Adelaxie  Dangerous Remove the problem urgently

Devil-67, 2020-07-19:
now I have tested with the V2.1 FCC.
All errors described above do not exist.
A2 is also available again.
BUT the sensor is lost and found again at regular intervals, it seems to me that the update time is too rare for A2, you can see it by the blinking asterisk in telemetry and, in my opinion, is therefore marked as lost.

Adelaxie, 2020-07-20:
@Devil-67 
So all problems are about LBT, right?
A2 is voltage value, a high rate of updates will be a waste.

Devil-67, 2020-07-20:
@Adelaxie  
Yes it only affects LBT 2.10
FCC 2.10 works without the problems I have described above.

If the sensor A2 is lost again and again because it is scanned too rarely, this is not a waste but necessary to ensure the function!
It does not have to be queried with the highest priority, but it seems to be necessary to query it a little more often.
There are also intermediate steps

Devil-67, 2020-07-21:
@Adelaxie the file cannot be downloaded, please correct it

Adelaxie, 2020-07-22:
what file?
[image]
this is a linked issue .

Devil-67, 2020-07-22:
Well, you close the issue, so I'm assuming you solved the problem.
then you use a file name with the extension ZIP and the name of the firmware.
What else should you think ...
If Frsky talked to her customers and didn't let them stand in the dark, it might go better!

Now I still don't know, has the bug been fixed or why is the issue closed?

RetoD7, 2020-07-22:
It was your own issue title with the ZIP filename which created the confusion. To have a clear view it doesn't help to have two issues open with the same problem.
As it seems others don't have this problem with the same firmware, it could help to provide more information to FrSky about your specific setup.
What S-Bus decoder is connected to the XSR?
What are your failsave settings?
Have you updated your X9D+ to OpenTX 2.3.9 or do you still use OpenTX 2.2.4?
(Have you tried to flash a new downloaded file, maybe your firmware file was damaged of an unknown way. Rebind after flashing was successful, might be on a new ID to prevent mistakes?)
(I'm not related to FrSky, I'm also just waiting until some issues are solved/fixed.)

Devil-67, 2020-07-22:
@RetoD7
If everything is ok with 2.10 FCC firmware, and from the same package the LBT has the problems, everything else doesn't matter what they said.
How many users use the XSR with 2.10 and LBT, how many know Github and that there is firmware for testing?

 if you have no problem, then keep you out please, it is not helpful to talk in between.

If Frsky needs more information they will let you know.

THANKS FOR YOUR UNDERSTANDING

Mike-HRC, 2020-07-22:
I am using V2.1.0 LBT on my QX 7 with a number of XSR (as well as R-XSR), and I don't get any issues with servos hitting end-stops - but I am using it on helis with a Flybarless controller.  

I don't use A2 or Channel generally so can't comment on those unfortunately, but the telemetry has been faultless on 2.1.0 (unlike 2.0.1) - I have probably 30 - 40 flights on XSRs with 2.1.0 LBT, no problems.

I didn't respond previously as I have nothing to help you - I only responded as you asked "How many users use the XSR with 2.10 and LBT, how many know Github and that there is firmware for testing?" - there's at least one more with a similar set-up to you.

Hope you get your issues fixed soon

Mike