FrSkyRC/Firmware-Test #87: S6R Receiver bindig not possible with ACCESS in D16 Mode

Skippydi, 2020-07-24:
Transmitter X10s with ACCESS Upgrade Kit, Firmware FCC 2.1.0 and OpenTX 2.3.9 NonEU
S6R Receiver Firmware FCC 2.1.0 2020-06-24

I tried to bind the receiver to the Horus X10s under ACCESS in D16 mode. Without success.
The red LED flashes no matter what I do
Various other receivers could be bound in D16 mode without problems.

P-I-Engineer, 2020-07-25:
i think there's a led issue with the firmware, mine flashes red when binding but is actually bound

Skippydi, 2020-07-25:
No, unfortunately not. I tried several times to bind the receiver. Without success. The red LED flashes without the receiver being connected.

P-I-Engineer, 2020-07-25:
Try hooking a servo up, the LEDs are all messed up with this firmware

P-I-Engineer, 2020-07-25:
Or connecting the Lua script.

I get no green leds

Skippydi, 2020-07-26:
Neither is successful, regardless of whether you connect a servo or try to connect to the LUA script.
However, the beta version of March 9, 2020 should work. Unfortunately it is no longer available on github.

mpjf01, 2020-07-31:
There's no D16 mode "under ACCESS", not sure what you mean by that.

There are a couple of beta version of the S6R firmware still available here. 
I'm assuming you tried the S6R version modified 10/3/20 - the release version from the FRSKY website in the package dated 24/6/20.
I have an X10S with the ISRM upgrade kit installed and the release firmware for both the S6R and the S8R allow a bind with it as normal.
Failure to bind normally is indicative of a firmware mismatch. I suggest that you reinstall the latest production 2.1.0 firmware on both your Tx and receiver. The fact that you can bind with other (presumably) 2.1.0 receivers suggests that it is likely an issue with the receiver firmware.

Skippydi, 2020-07-31:
Of course there is the D16 mode under ACCESS. ISRM D16.
I can bind all other receivers with ACCST in D16 mode under ACCESS, but not the S6R.
I have tried different firmware versions. Nobody works. The firmware version on all devices is FCC, that is secured.

mpjf01, 2020-07-31:
As I wrote there's no D16 mode "under ACCESS". D16 mode is ACCST, a different protocol entirely. The X10S with the ISRM upgrade module installed supports both the ACCESS and ACCST protocols. 

If you have tried the firmware versions that I mentioned in my last post, and then re-flashed and tried again, and still the receiver won't bind, the only conclusion remaining is that the receiver is faulty.

Skippydi, 2020-07-31:
The only strange thing is that under ACCST the receiver can be connected to a transmitter as usual and also works. I therefore rule out that the receiver is defective.
But I will flash the firmware in the transmitter again to make sure the error is not there.

Skippydi, 2020-07-31:
I flashed the receiver again with the firmware from June 24th, 2020. With my Horus X10. Now I can bind the recipient and it works.

closed 2020-07-31 by Skippydi