FrSkyRC/Firmware-Test #89: No CPPM signal on X4R v2.1.0 EU/LBT ACCST

abenn1, 2020-07-27:
Having just updated my EU/LBT X4R ACCST receiver (white wrapper) to firmware v2.1.0 from the FrSky download site, I no longer have CPPM output.  I have read the thread https://github.com/FrSkyRC/Firmware-Test/issues/15, but that is closed so I must raise this as a new issue.

The receiver has been working with X4RSP_EU_170322.frk firmware (or its predecessor) in CPPM mode in a fbl heli for the past two years without problem, but when I updated it to v2.1.0 my BeastX fbl controller reported "No receiver".

I have carried out the following basic checks on the receiver today:-

1.  Re-flash it back to v1 X4RSP_EU_170322.frk firmware and bind it to my v1 Taranis X9D+ Tx with signal pins 2 and 3 connected by a jumper, and I then have CPPM output on channel 1 as before, and no PWM output on channels 2, 3, and 4.  I believe this is how it should be.
2.  Re-flash it to v2.1.0 ACCST LBT firmware from the FrSky download site, and bind it to my v2 Taranis X9D+2019 Tx with signal pins 2 and 3 connected, and I have no CPPM output, but I do have PWM output on all four channels.
3.  Re-flash it to X4R_CPPM_ACCST_2.1.0_LBT.frk firmware from the Github page, and the result is the same as with the FrSky v2.1.0 -- i.e. no CPPM output, but PWM on all four channels.

agetommy, 2020-08-07:
With FW 2.1.0 you must bind the X4R in "Ch9-C16" mode to get a CPPM output an Ch-Out 1.
Bridging 2+3 is not necessary and without an effect.

robustini, 2020-08-27:
The PPM signal is there, but it is somewhat incompatible with some systems, where for example that of a D4R-II is perfect.

abenn1, 2020-08-27:
OK, thanks.  I'm still confused though, for with v1 firmware it had worked fine with my MicroBeast heli controller, and with the X4R_CPPM_ACCST_2.1.0_LBT.frk firmware it also works.  But with the regular v2.1.0 ACCST LBT firmware from the FrSky site, the MicroBeast doesn't recognise it.  My previous comment about different binding process actually didn't work.  Why should the CPPM signal be different between those three versions?

Adelaxie, 2020-11-03:
Hi @abenn1 
Sorry for your inconvenience.
Since the product has been discontinued for a long time, some special versions of functions might not be considered when we released the update of 2.1.0.  
Please use X4R_CPPM_ACCST_2.1.0_LBT.frk for your situation.thanks