FrSkyRC/Firmware-Test #91: crash with G-RX ACCESS 2.1.0 opentx 2.3.9 and Xlite pro 2.1.0

Djodjotje, 2020-08-02:
[image]
[crash.zip](https://github.com/FrSkyRC/Firmware-Test/files/5012080/crash.zip)

Hello, 

I am a starting DLG pilot and do not throw high yet, but have crashed.

A telemetry reset may affect the elevator and ailerons for a short time.

I did a telemetry reset on switch SE to get the G-RX6 to the correct height, this is also my start button. I release this button when it is at its highest point. After this he shot down like a rocket. I have my elevator and ailerons at a value of 50% in openx, so I find it strange that values rise to 1024.

Can anyone say something useful about this crash moment. ???

Thanks

Allerhopp, 2020-08-02:
What you see in the logfile is the gimbal data. It seems to be an issue with the right gimbal (if you fly mode 2?), aileron and elevator are affected. A loose connection eventually. It is a hardware thing imo. Yellow is SE action, red is the suspicious gimbal data.
[image]

RetoD7, 2020-08-02:
Sorry that's the wrong place for your problem!
Telemetry reset is a function of OpenTX and has nothing to do with the tx module (ISRM) or G-RX6.
The recorded control values are the inputs of your TX. If you have not moved your right stick from one edge to the other the hardware of your Xlite is faulty.

Djodjotje, 2020-08-02:
sorry i thought it might have something to do with the G-RX6 firmware,