FrSkyRC/Firmware-Test #92: Archer GR8: CH2 + CH3 unwanted servo movement at startup

RetoD7, 2020-08-20:
When powering on the Archer GR8 (ACCESS 2.1.0) the the channels or more exactly pins 2 + 3 are generating some pulses around 200kHz. It's similar to the well known startup movement of some servos. But the error in the GR8 causes a little bit later a second additional movement.

RetoD7, 2020-09-09:
The **Archer R8 Pro** (ACCESS 2.1.0) has exactly the same Error. **The risk of some damage to the servo or plane is quite big.**
It seems all Archer RX have this problem, as all use the same firmware?

EngelMT, 2020-09-17:
I have a R8 Pro on the table for other tests. So it was easy to make the video from switch on and off the receiver with several different servos connect. I can not find any servo movement after switch on.
[R8 Pro.zip](https://github.com/FrSkyRC/Firmware-Test/files/5238224/R8.Pro.zip)

RetoD7, 2020-09-17:
Digital servos seems not to be affected. (And I hope also ESC's are not affected, because many have them on CH3.)
But all tested **analog servos** (Futaba, Hitec, Waypoint, ..) make about ** 1/3 up to 1/2 of their full way**.
Only CH2 and CH3 are affected. If the servo is connected to an other channel, there is no movement. (Or depending on type just their own usual power up movement.) 
The pulses of about 200kHz are easy to measure with an oscillograph, so there is a disturbing signal without any purpose.

EngelMT, 2020-09-17:
O.k. so FrSky has to check it. I think Adela will read it here. I will inform Adela again separately.

Adelaxie, 2020-09-23:
Will have it fixed along with the next update.

RetoD7, 2020-10-01:
Problem is solved with the new firmware Archer-X_2.1.2.

closed 2020-10-01 by RetoD7