FrSkyRC/Firmware-Test #93: X9 Lite S WARNING TX antenna problem!

maseaua, 2020-08-28:
Hello,
Hardware X9 Lite S, 
OpenTx: 2.3.9-frsky, 
Int. Module: ISRM-X9LiteS , 1.1.1/2.1.0 EU

I'm having this issue where the radio would throw a "WARNING TX antenna problem!" message when I wave my hand in front of the antenna (while the antenna is bent left at 90 degrees or when straight I need to use 2 hands top an bottom of the antenna). I posted a video on YouTube with the issue: https://www.youtube.com/watch?v=PwLPK8yHDcI

I did not check for this issue before updating the internal module ISMR-S-X9Lites from 1.1.3 to 2.1.0 EU. I've read on some other forums in the past that this might be a firmware issue, I'm wondering if this might be the case?

Best regards

Gozzz, 2020-08-28:
Mine does the same w/2.1.0 fcc.

maseaua, 2020-08-28:
Do you by chance have the initial version 1.1.3 ? maybe the issue is not reproducible.

Gozzz, 2020-08-28:
> Do you by chance have the initial version 1.1.3 ? maybe the issue is not reproducible.

I do not.

Gozzz, 2020-08-28:
Also, when my antenna is bent to the left my RAS hovers around 35-40, and when it is straight RAS is 0. I usually fly with the antenna straight now for that reason.

It is also much worse when I have the xjtlite module plugged in, so I usually remove that when flying with the internal isrm module.

maseaua, 2020-08-28:
Thank you for the input! I am not sure if the initial version which came with the device was 1.1.3, could you please confirm this?
Thanks! 
Can this be a software issue?

maseaua, 2020-09-01:
Indeed this TX antenna problem, it's because the new software 2.1.0. I've compared the range with my other X9D 2019 using same internal module version 2.1.0 , and the range is the same on both: worse.

Gozzz, 2020-09-23:
> Indeed this TX antenna problem, it's because the new software 2.1.0. I've compared the range with my other X9D 2019 using same internal module version 2.1.0 , and the range is the same on both: worse.
> 
> Switching back to 1.1.3 on X9 Lite S solved all my problems, SWR is 15 with the antenna bent to the left and 0 when straight.

Can you message me where you found 1.1.3 for the x9 lite s? I've had range issues where clearly there should not be.

maseaua, 2020-09-23:
I've e-mailed the technical support department from Fr-Sky -> sales4tech@gmail.com, detailed the issue and asked for 1.1.3 firmware. They kindly provided the firmware along with other instructions for problem investigation. Answer was quick, ~1 day.

If you get no response, I would gladly help.

Adelaxie, 2020-09-30:
Will fix this issue asap on 2.1.x, sorry for your inconvenience

Gozzz, 2020-12-31:
> Will fix this issue asap on 2.1.x, sorry for your inconvenience

How's this update coming along?

Flaps1978, 2021-01-09:
I don't have much to add except I am seeing the same problem on my X9 Lite S.  Upgrade to 2.1.0 FCC and I start getting random TX antenna warnings, downgrade back to 1.1.3 FCC and all is well.  It's stopping me from using a new Archer RS that I bought not long ago.