FrSkyRC/Firmware-Test #97: Archer R10 pro, FW Archer-X_2.1.2: showing 2.1.0 after flash, servo movement at startup

pau33, 2020-10-04:
x9d+2019 SE, Opentx 2.3.9 LBT, new Archer R10 Pro

I flashed the the Archer-X_2.1.2.frsk from FrSky Homepage via both ways several times, OTA and via STK. Both modes show ok at the end of flash.

But always 2.1.0 ist shown, If I would try to flash per OTA again. (Is there another way to show receivers FW-Nr. in opentx?)

I connected two analog servos to several PPM outputs. They all moved to their endpoints at startup firstly (with various Failsafe-settings). That ist not acceptable in a model use.

All in all that behavior would be explainable, if it is not the 2.1.2 on the FrSky-Server. But that would be to easy, wouldn't it?

pau33, 2020-10-04:
I now did the same test with a new Archer R8 Pro. It was originally on 2.1.0 because ist is a new one.
Same procedure as above, same result (shows 2.1.0 and analog-servo-movement, it is the same Archer-X_2.1.2.frsk).

The analog servo is a Graupner C261.

It additionally plugged in a digital servo (Graupner DES 488 BB MG) on both, R8 and R10. This does not show up a movement on startup on both receivers. So may be, this movement can also be due to analog/digital or the servo itself.
Is there a confirmation for this observation?

EngelMT, 2020-10-05:
I think Adela will asnwer after she is back from National holidays. In the moment nobody is in the factory.

Adelaxie, 2020-10-10:
> I flashed the the Archer-X_2.1.2.frsk from FrSky Homepage via both ways several times, OTA and via STK. Both modes show ok at the end of flash.
But always 2.1.0 ist shown, If I would try to flash per OTA again. (Is there another way to show receivers FW-Nr. in opentx?)

Re-register and bind would have the correct version showed but we will fix it in the next release at the same time. thanks

Adelaxie, 2020-10-10:
> So may be, this movement can also be due to analog/digital as it is the same for two analog servos (C261 and FB-S35S).
Is there a confirmation for this observation?

Let me confirm it with our testers, it seems like a pulse when powering up from your description.

gvt70, 2020-10-10:
I have the same problem on the version 1 16 channel redundancy buss. Everytime the RB is turned on there seems to be a signal spike to the servo's that drives them some degrees. Then when the RX led turns green, they go to their set neutral point again.
I tried multiple powersupplies, they all do the same.
It even happens when there is NO rx connected to the RB and only power is supplied. But then of course the servo's turn and stay at that position because there is no rx signal to return them to neutral point.

There should not be any signal to servo's before RX has solid connection to TX (led is green)!!

steand, 2020-10-12:
> Re-register and bind would have the correct version showed but we will fix it in the next release at the same time. thanks

When the receiver is installed deep in the model it isn't be able to re-register the RX, easily. 
My requirement would be that after the update the right version of RX viewed in the TX correctly.

Hobby4life, 2020-10-19:
Please fix both asap, Re-registering is almost not possible is some cases.

The servo "jump"  issue is even more important.. it can cause mechanical faillures in models.

Adelaxie, 2020-11-03:
Please check the latest 2.1.5 which fixed both issues.

pau33, 2020-11-07:
I flashed the R8Pro and the the R10Pro OTA with the 2.1.5 Firmware and repeated the tests with all 3 servos from above.
The results are:
- OTA flashing works. But the communication with 2.1.5 does not recover after 5s, as posted in the readme with 2.1.5. I had to recycle the receivers as before.
- The correct firmware 2.1.5 is now shown without rebinding. This issue is fixed.
- The servo movements (max. with the analog servos an no with the digital servo) occurs as before. The behavior has not changed when powering on. That may be still due to analog/digital nature of the servos. But I tend to say, the receivers have an impact, because when I plug in the servos to a receiver, which is already powered on, there is only very little movement on the analog servos (and still no movement on the digital servo).

Adelaxie, 2020-11-10:
>But the communication with 2.1.5 does not recover after 5s, as posted in the readme with 2.1.5. 
Have you tried a second time? OTA update has the possibility to break by some reasons.

The servo movements (max. with the analog servos an no with the digital servo) occurs as before. The behavior has not changed when powering on. That may be still due to analog/digital nature of the servos. But I tend to say, the receivers have an impact, because when I plug in the servos to a receiver, which is already powered on, there is only very little movement on the analog servos (and still no movement on the digital servo).
>Its fixed in bootloader so only in new units. Please contact with after-sale support if the problem is annoying you.

Hobby4life, 2020-11-10:
Updating to v2.1.5 on a R10 Pro receiver caused smartport telemetry to stop.
**No** communication with sensors (neuron80, gps)

Flashing the R10 back to v2.1.0 all worked fine again.

pau33, 2020-11-10:
> after-sale support
@Adelaxie:
Do you have an email for the after-sale support for me?
Or an instruction, how to flash the bootloader of the R8/R10 Pro?

@Hobby4life:
I added an SM-GPS-Logger as a sensor to the S-Port of the R10Pro and at a first look it seems as if the S-Port ist working with my 2.1.5 firmware. I can see the telemetry values.

Adelaxie, 2020-12-04:
@pau33 check the receiver options after bound, if f.port was selected . thanks.

pau33, 2020-12-07:
@Adelaxie:
F-Port is not selected. I need the S-Port for the GPS-Logger.