FrSkyRC/Firmware-Test #100: ability to use the RSSI on channel 16 or 8

notlebowski, 2020-10-17:
Flight controllers can create a virtual S-Porr sensor that's based on the channel 8 or 16 output of an XM+ receiver.

The G-RX8 nor other receivers with an S-Bus input are capable of that.

It would be very useful to measure the RSSI of the secondary receiver to make sure that in some situations you can rely on it.

Adelaxie, 2020-11-03:
@notlebowski thanks for your advice!
Only receivers might be used for drones now has the feature for now.
What kind of plane you are using on with G-RX8?
Are you meaning the flight controller for airplane?

notlebowski, 2020-11-03:
I'm flying a hotliner with that G-RX8
I know that drones have that ability. I just want a G-RX8 have that ability as well.

If my main receiver has no reception I want to rely on a good reception by the slave receiver.
The 2nd receiver may have poor reception all the time and be of no use and I would have no clue as I don't have any RSSI information at all of that secondary receiver.
But it is not working

notlebowski, 2021-01-13:
I tried to do this by using the analog RSSI output of the XM+ 
When actually implementing it I found out that this RSSI-output does not work....   Nobody ever had that working.