FrSkyRC/Firmware-Test #103: Telemetry problem X9Lite/S8R ACCST 2.1.0

PsykopathCZ, 2020-11-09:
I have got quite a problem with telemetry. I have got X9Lite radio and S8R receiver, as telemetry I am using FAS40 current sensor and Vario normal precision. When I am using FW version 2.1.0 - latest on FrSky pages - for receiver and radio, i sometimes get wrong values for each sensor. For example -8000000 Volts, 43000 Amps, 20000000 meters of altitude and so on (you can see them on attached screen from companion). These values are not prenament, but few times in fligt I get them.

If I downgrade firmwares to older - radio version ISRM_N_190805 and S8R version S8R_LBT_20190531, values are correct.

Is it any bug, or I am doing something wrong?
[image]

Nerenye, 2020-11-19:
It's a big. Please look and comment to the other issues

CBMSOLRAC, 2020-11-27:
@PsykopathCZ, Hello do not give them more laps, it is a problem of the SxR firmware, look at my open problem 106
Luckily it only affects telemetry data, and not normal operation.
Let's see if Frsky finally solves this problem, which increases with the internal ISRM module of ACCESS radios, and which Adela knows perfectly.

Nerenye, 2020-12-25:
I don´t think its a SxR fw bug since we are using grx6 /8