FrSkyRC/Firmware-Test #104: R-XSR Accst LBT f-port not working with brain

Dracks, 2020-11-10:
I received today a new R-XSR, I flashed with the f-port version of the firmware, but when I try to connect to the mini-brain, it restarts every seccond, and don't connect to the radio. If I start the transmitter, without plug to the mini brain to f-port, it starts and binds correctly to the radio. 

I've found this thread from brain, that seems the problem is the f-port implementation of the r-xsr:
https://www.msh-electronics.com/forum/general-discussions/r-xsr-f-port-brain-2-telemetry/

Adelaxie, 2020-12-08:
@Dracks Can you share a photo of your connection in very details? especially about RXSR end.

Dracks, 2020-12-11:
Hi,

Sorry, doing some tests I plugged it wrongly, and I burned it up. Did you know why the rx4r doesn't have an fport version of the firmware, but the x4r, had it? 

Thanks,

closed 2020-12-11 by Dracks