FrSkyRC/Firmware-Test #106: ISRM MODULE PROBLEM FOR UPDATED X10S TO ACCESS AND D16

CBMSOLRAC, 2020-11-23:
ISRM MODULE PROBLEM FOR UPDATED X10S TO ACCESS AND D16

Hello fellow hobbyists

I have a problem and after several tests I am unable to find the reason or the solution and all this happens after mounting the new ISRM module

I indicate the material used and the firmware that are installed

Radio: X10S
RF module: new ISRM module for updated X10S to ACCESS and D16 ACCEST purchased 15 days ago in Engel MOdellbau
Receiver: tested S6R and S8R
Radio firmware 2.3.10 with "internalaccess" selected in Companion and installed
ISRM module firmware: ISRM-S-X10S_2.1.0.frsk CRC: C2C91E1C (latest version available)

Process:
1st Downloaded and installed 2.3.10 on the radio, then updated the ISRM module with the indicated firmware. All right
2nd new Bind of the receivers selecting ISRM D16 mode with receiver number one with 5 and the other with 8
3º All sensors erased and detected again without problems
The S6R with a MLVSS for battery voltage configured with “cell” and 3 elements
The S8R with a FLVSS and a variometer (configuration same as the other but Variometer height

The RSSI on the two receivers is good and maybe somewhat higher than with the original IXJT module.

The problem is that up to 8-10 meters everything works fine and from that moment the measurement of the battery in the S6R and the battery and variometer in the S8R begin to give wrong values ​​and the programmed low battery alarms appear.

I have configured the reading of minimum values ​​for "cels" (Battery) and for example, 4S (16V) gives me in several readings 8, 9, 7, 16,16,11 etc, with the variometer values ​​of 800 meters, 13000 meters etc
 On the contrary, the direct RSSI values ​​of the receiver always work well, it is as if there were reading problems with the "S.Port" and the new ISRM module.

I would appreciate your help, or what I can do since I have all the planes at the moment in stanby
Apologies for my translation

A hug
C. Bosch

CBMSOLRAC, 2020-11-24:
I have carried out more tests with the X8R receiver and the problem disappears, I understand that it is a problem with the S6R and S8R receivers that already gave problems with telemetry parameters and refresh rate.
I need a solution since I cannot use the SxR with the ACCESS module and D16
Regards

EngelMT, 2020-11-24:
@CBMSOLRAC  I think it is the known telemetrie issue you reported. Internal sensors are working well. But external ones shwo from time to time extreme values which do not correspond to reality. This problem occurs more frequently with some recipients, less frequently with others. The SR Series is particularly affected.

The transmission of control signals is not affected. It is only a matter of telemetry.

FrSky had announced to work on the problem. So it should be fixed with a new update.

By the way, the problem already existed with ACCST.

CBMSOLRAC, 2020-11-24:
**@EngelMT** Thanks Andreas for the answer
After several tests, I have come to the same conclusion it is a firmware problem of the SxR family, the latest version was still in testing and a final 100% revised version has not been provided.
I understand that Frsky has other priorities, but I think he has had more than enough time to finish this topic.
Although it only affects telemetry, keep in mind that not having reliable values ​​on the status of sensors, for example, battery can cause model breaks and mistrust, as well as problems with all new radios in D16 version
Finally I have loaded a firmware version 2.1.0 LBT version of March 2020, and the issue now works without problems, today I have been flying 2 more models of 6 flights and without any telemetry alarm.
If the problem was also with ACCEST, I don't know but with the IXJT module it did not present this defect, this has appeared with the New ISRM Module
For now I will continue operating with this version of Firmware until FRSKY wants to keep its users happy and in my case for more than 10 years
Greetings to Andreas and his magnificent team