FrSkyRC/Firmware-Test #110: Archer RS, slow Sport telemetry refresh for extra sensors when used as redundant Rx

DaySpeed955i, 2021-01-18:
Sport is selected.
Latest firmware.
X12S with ISRM from factory, ISRM-S-X10E, 1.1.0/2.1.0FCC.
OpenTx 2.3.11

Archer RS works very well alone, control and telemetry for all extra sensors are fine, including RB-20.

When Archer RS is used as redundant Rx, control is always good. Telemetry is good from primary Rx(SR10 Pro) as long as RF is from primary. When RF is transferred to redundant Rx(Archer RS), telemetry refresh rate for extra sensors become erratic, it can take up to 4 seconds to refresh extra sensor values. When RF is transferred back to primary Rx, telemetry refresh rate for extra sensors stay erratic, when Sport connector from Archer RS is disconnected, telemetry for extra sensors is back to normal refresh rate, when Sport connector from Archer RS is reconnected, refresh rate become erratic again.

There is no telemetry lost.
Rx sensors RSSI, VFR, A2, RxBt are always normal.

Tested with 2 SR10 Pro and 2 RS with same results.

When RX6R(RX6R_ACCESS_2.1.0.frsk) is used as redundant Rx with SR10 Pro as primary, there is no telemetry refresh problems.
When SR10 Pro is used as redundant Rx with SR10 Pro as primary, there is no telemetry refresh problems.

SmartTommy, 2021-03-11:
Hi DaySpeed955i,

have you try update your firmware to 2.1.6?  
[https://www.frsky-rc.com/archer-sr10-pro/](url)
[https://www.frsky-rc.com/archer-rs/](url)

could you show me a video?

DaySpeed955i, 2021-03-14:
Hi,
I made a short video, hopefully enough information in it, I have a long video but too big to post here.
X12S ISRM-S-X10E v2.1.6FCC
Archer RS v2.1.7
Archer SR10 Pro v2.1.6

https://user-images.githubusercontent.com/62478798/111056155-e9ce7480-844a-11eb-80e6-cb22a77f3e55.mp4

DaySpeed955i, 2021-04-19:
Solved with v2.1.8.

Thanks!

closed 2021-04-19 by DaySpeed955i