FrSkyRC/Firmware-Test #114: The receiver does not bind. r9m + r9mm-ota

ONESucH, 2021-02-10:
There is an old r9m receiver there is an old taranis x9d + remote control there is (access).
I tried to flash a transmitter with checkbox support "flexr9m" openTx 2.3.12, Bootloader - up to the latest version, downloaded a fresh SDCard to a USB flash drive and put it in taranis.
I flashed an external module X9R for access FLEX (F.Port l S.Port).
I flashed the external module r9mm-ota for access FLEX (F.Port l S.Port)
When I tried to flash the receiver, it was not flashed the first time, perhaps because of the old device, but r9mm-ota was flashed successfully.
I know how to bind, I held the bind button in the receiver when the remote was listening to the receivers, after holding and listening to it, I gave power to the receiver, pressed Enter & exit to complete bind, rebooted and there was no bind.
I tried to connect everything in different ways and did not come to anything, I have 2 r9mm-ota receivers both behave the same, no bind happens.
Can you send the archives of the necessary firmware for the bind to be successful?
I will send a photo of my device, receivers, manuals, they tell me that links are cut from off sites for old versions, I tried everything that is on the site but nothing fits.

In the remote control, in the model settings, I turned off the internal module, turned on the external r9m, 868 Hz, 16 channels, telemetry-on, I am powered by the battery from the bf4 flight pilot.
I just can't get it to start moving the stick, you don't have anything about it in the manual, I don't know what to do and no one can help me with this question, I only turn to you with the hope of finding out what's
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

RetoD7, 2021-02-11:
**This is not an issue.** A forum would be a better place for such problems / questions.

Your R9M does not support ACCESS, but the newer R9M 2019 would. You find the correct firmware under "Discontinued Products".

ONESucH, 2021-02-11:
can you provide links so that I can download right away and not be mistaken with the firmware for r9m + r9mm-ota + taranis x9d+?

ONESucH, 2021-02-11:
I downloaded from these links and it didn't help.
screen - https://prnt.sc/z41drb

RetoD7, 2021-02-11:
If you don't know the official website of the manufacturer, you are definitely wrong at this place.
https://www.frsky-rc.com/product/r9m/
https://www.frsky-rc.com/product-category/receivers/900mhz-r9/

**Please close this Issue.**

ONESucH, 2021-02-11:
I wrote that I already downloaded the archives bind does not work from these archives.