FrSkyRC/Firmware-Test #116: R9MM(OTA) would show no stick move when channel3 < -100% (950us)

smaller09, 2021-03-06:
I have r9m 2019 and 2 r9mm-ota and 1 r9mm all flash to ACCESS firmware.  all connected to FC by F.port.
And found that if the channel 3 range bigger than -100~+100 on transmiter, and when THR stick move to the lowest. the receiver would on send out the rssi to FC, and the FC will show  no stick movement of transmiter. but telemetry always works correct at that time.
I had tried downgrade the firmware of r9m & r9mm to the lowest ACCESS firmware and ACCST firmware. the problem was still there. change sbus output also not solve the problem. the only way to run out of this is reduce the channel range of channel3.
I did not test other channel .but I thing if channel 1 or 2 lower than -100% the problem will always be there.

Adelaxie, 2021-05-18:
what is your radio firmware and version please?

SmartTommy, 2021-05-25:
> I have r9m 2019 and 2 r9mm-ota and 1 r9mm all flash to ACCESS firmware. all connected to FC by F.port.
> And found that if the channel 3 range bigger than -100~+100 on transmiter, and when THR stick [...]

[image]
please set the FC channel range like the image,thanks

Lagarda, 2021-06-11:
What frimware did you use?