FrSkyRC/Firmware-Test #117: RXSR not always powering up as redundant Rx.

DaySpeed955i, 2021-03-14:
X12S ISRM-S-X10E V2.1.6FCC
SR10 Pro V2.1.6
RXSR v2.1.0 ACCESS

If RXSR is used as primary, it is always working well.

When RXSR is used as a redundant Rx and Sport is connected, RXSR is not always powering up, since primary Rx is working normally, we might not have redundancy. Tested with 2 sets of RXSR and SR10 Pro.
If Sport from RXSR is not connected, RXSR is always powering normally.

Note that when the RXSR in not powering up, the Sport signal is not normal, signal is from 1.2v to 3.6v instead of ground to 3.6v.

Also, RXSR have the same problem as Archer RS for refresh of extra telemetry sensors as seen in issue #110 

https://user-images.githubusercontent.com/62478798/111087314-5606b200-84f7-11eb-9f46-4b98f2a823a1.mp4

Devil-67, 2021-04-09:
I just can approve that too!!!
A few other Frsky receivers are affected with the same problem.
Another big problem is that RSSI and VFR are not displayed in telemetry by both receivers at the same time.
This leads to the fact that if you install the RX hidden in the fuselage you do not notice that it is not working.
So a real Frsky problem, they should fix it soon!

Here is a summary of the receivers I've tested. 

here the updated summary version:
I don't have more ACCESS capable RX, but that should be enough.

last status from April 5th, 2021:

Everything connected means:

1. Plus with plus

2. Minus with minus

3. S-Port with S-Port

4. S-Bus out with S-Bus in

Working setups:

RXSR and RXSR in redundancy operation everything directly connected to each other works and the receivers boot normally.

G-RX8 and RX6R in redundancy mode everything directly connected to each other works and the receivers boot normally.

Archer GR8 with RS in redundancy operation everything directly connected to each other works and the receiver boots normally.

Archer SR10 Pro with RS in redundancy operation everything directly connected to each other works and the receiver boots normally.

Archer SR10 Pro with RX6R in redundancy mode everything directly connected to each other works and the receiver boots normally.

New category:

Works very often but not always

Archer SR10 Pro with RXSR in redundancy mode everything works directly connected to each other and the receivers often boot normally, but it also happens from time to time that the slave does not boot, but it is quickly overlooked that it does not boot and therefore still useless because you also via telemetry does not see that the RXSR may not have booted this time and therefore I put it in the faulty category.

Incorrect setups:

G-RX8 with RXSR in redundancy mode, everything is directly connected to each other, the RXSR only boots sporadically every time it is switched on.

G-RX8 with Arche RS in redundancy operation, everything is directly connected to each other, the RS only boots sporadically every time it is switched on.

Archer Gr8 with RXSR in redundancy mode, everything is directly connected to each other, the RXSR only boots sporadically with every Xth power-on.

Archer SR10 Pro with RXSR in redundancy mode, everything is directly connected to each other Often, but unfortunately not always boots reliably.

Here is another combination that is completely out of line:

Archer RS ​​and with RXSR in redundancy operation, everything is directly connected to each other, the RXSR does not boot at all in this case and the external S-port of the Archer RS ​​fails completely, only the internal sensors RSSI, Rxbt, VFR of the RS still work.

Here, how the RX are connected to each other, more is not necessary to test this, but I also had S-Bus In connected to S-Bus Out, as it is actually used in the model.

The illustration is from the Doc from Jet Flyer linked above. 

[image]

DaySpeed955i, 2021-04-19:
Solved with v2.1.7.
Thanks to Andrea and FrSky team!

closed 2021-04-19 by DaySpeed955i