FrSkyRC/Firmware-Test #118: Telemetry not working with R-XSR ACCST D16 v2.1.1(F.Port firmware update only )

klutvott123, 2021-04-03:
There has been several reports of telemetry not working with the R-XSR ACCST D16 v2.1.1 F.Port firmware. 
Telemetry works with 2.1.0 but not with 2.1.1. 

https://github.com/betaflight/betaflight/issues/10496
https://github.com/betaflight/betaflight/issues/10591
https://github.com/betaflight/betaflight/pull/10656
https://www.rcgroups.com/forums/showthread.php?3812029-Last-update-R-XSR-FPORT-2-1-1-LTB-Telemetry-broken

daleckystepan, 2021-05-03:
Does anybody care?

klutvott123, 2021-05-03:
I care and Frsky should care. Right after opening this issue I sent Frsky an email with links to this and the issues. That was a month ago and still no response. @Adelaxie?

daleckystepan, 2021-05-03:
Sure. I meant someone from FrSky. I have reported this bug also through my local reseller but without any success.

bob01, 2021-05-06:
++

v2.1.1 f.port
- flight controller (betaflight 4.2.9) sensors can only discoverable for about 20 seconds after power on
- rx sensors (eg rssi) are fine
- activity on FC sensors (device Id 245 I think) slowly decreases to nothing over about 10 seconds followed by voice warning ‘sensor lost’ after which all FC sensors turn red and activity ceases until next rx power cycle

Reproduced with...
-  3 R-XSR receivers (all new), flashed to 2.1.1
- 2 Aikon F7 v2 flight controllers w/ betaflight 4.2.9

- v2.1.0 subs / s.port - works
- v2.1.0 f.port - works
- v2.1.1 f.port - FC telemetry degrades over ~10s then dies completely

Strange thing is - friend bought the same gear (radio, everything at the same time), flashed the same FW on all components - his works
Betaflight dumps compared, no significant differences

haslinghuis, 2021-05-14:
[image]

klutvott123, 2021-05-14:
So how do we reach Frsky? Not answering here and not answering emails. The Frsky product manager is watching this repository so I assume she's getting the notifications.

Adelaxie, 2021-05-21:
[RXSR-RXSR-FC-fport-beta-0521.zip](https://github.com/FrSkyRC/Firmware-Test/files/6520836/RXSR-RXSR-FC-fport-beta-0521.zip)

Can you help to confirm this fix?
tested on three different FCs here .

daleckystepan, 2021-05-22:
Tested ACCST LBT FPORT and seems to be OK at least at home. Flight test later today or tomorrow .

TheIsotopes, 2021-05-22:
@Adelaxie Tested also Accst LBT FPort on the bench and everything seems to be working again. 
a flight test is not expected at the moment, just too windy.

Tommy032017, 2021-05-22:
@Adelaxie Tahnk you.
I also tested RXSR_Accst_Fport_LBT_2.1.2-beta.frsk and everything seems to work perfect - no more issue using RX + GPS and telemetry lost. Configuration RXSR - MATEK F411 Wing, F.port TX1 (UART1), GPS UART2, soft serial off. Real flight still pending.

daleckystepan, 2021-05-22:
6 flights today without telemetry lost issue. LGTM

Adelaxie, 2021-05-24:
thanks.Will update website after more tests.

closed 2021-05-24 by Adelaxie

daleckystepan, 2021-06-02:
@Adelaxie Sad news. My quad crashed probably due to RX issue with the BETA 0521 FW provided by you. Be aware before publishing this FW on your websites.

Have you noticed any issue with this FW during more testing?

[image]

TheIsotopes, 2021-06-19:
so what is it now? is the beta from above stable enough for daily use or will there be another beta?

ProfDrYoMan, 2021-06-22:
I hope that this get's open up again, with some followup FW eventually.

TheIsotopes, 2021-07-13:
ok, for me the subject is over here. One seems to be waiting in vain for an answer.
I have downgrade my receivers to the old firmware accst v1 again. 
everything is going perfectly and there are no problems.

daleckystepan, 2021-07-13:
Gived up. Bought IRC Ghost.

jesuslg123, 2021-09-14:
Hello! I have recently switched my R-XSR from SBUS to F.Port and I am having this problem. I can discover sensors and have telemetry updates during the first seconds after boot the FC, but after a few seconds the sensors stop getting new information.

Is there any update regarding the issue? Any way to get older 2.1.0 or similar versions to try out? 

Thanks!