FrSkyRC/Firmware-Test #119: S8R Blue LED not Working

szempruw, 2021-05-10:
I am a "normal" FrSky User. I updated old X9D Tx and older X9D+ (pre 2019) using Companion 2.3.11. I read about need to update FrSky receiver software to 2.1.x Software. I updated internal module in Tx to XJT 2.1 off the FrSky website. Also updated all my Rx's to the latest 2.1 .x software and specifically for the S8R Rx using SXR_ACCST_2.1.0_FCC.frk.

The update was flashed successfully using the Tx FIRMWARE directory, with firmware loaded on the SD Card and flashed to external device connected to the Rx Smart port.

I bound the Rx using XJT D16, 1-16 Ch and Telemetry On and then used the scripts copied into the SD Card to calibrate the receiver. Calibration successful. (Thumbs up) Using the S8R setup lua script, I chose orientation and enabled both SxR function and Quick Mode. I left all other values alone.

I installed Rx into a plane. I tried the 3-pos switch on Channel 12 to attempt the Self Check routine. The Blue LED did not come on. I next tried to press the F/S button on the Rx and there was no Blue LED or servo twitch as I expected. I moved the other joystick function as indicated and then the throttle servo. The plane's functions responded correctly to movement as expected. The assigned Channel 10 switch turns the QM off to Stabilize to Auto-level as expected. The software seems to work, just no confirmation or indication of Self Check routine working and no temporary Blue LED when first turning the Rx on

I have two S8R's and both will not turn Blue LED on. I only get the yellow, red and green lights as expected, green to indicate bind ok.

I reference RCDIY Oscar Liang video for correct setup of an S8R as I know it as a "normal" user. http://rcdiy.ca/sxr-setup-s6r-s8r/

I don't know if this Blue Light issue is FrSky Rx 2.1.x or Lua Script or OpenTx software related. It just doesn't seem to work as expected.

Kindly, WTS

szempruw, 2021-05-10:
I followed User instructions.  I know how Self Check is supposed to work. The throttle and trim tab was set to lowest position (-100) and not touched.  No blue led by switch or by pressing F/S on receiver. No temporary Blue when receiver turned on.

DaySpeed955i, 2021-05-10:
Try reducing down to 0 to test if it works the following value:
 SYS/Radio Setup/Play delay

It solved my problem.

szempruw, 2021-05-10:
OK, 1/2 apology needed. I added Thr output value to -105% travel, checked the travel, and I got the switch to turn on blue light. I moved the sticks and after twitching the green LED came on.  Sometimes the Self check quit too soon. After 3 joystick moves Ail, Ele, Rud and the twitch the green LED came on. Then I bought Thr to neutral. I tried the F/S press with Thr -105% because you are supposed to be capable of performing a self check in that way. (either switch or F/S press)  It would not work. I disconnect battery and re-connect and no temporary blue light when Rx turned on. So I am only half successful but I still don't think it is reacting right. When you try many things at once, you lose track. I am going to re-check Thr lockout again and order of calibrating in Horz Normal, then change it to upside down Horizontal Reverse then Self Check being able to change to check as installed and servo reverse in Lua or reverse in Tx Output. I'm sure it must be done in exact order.  Lua script slow to load and recognize Rx status. Noticed some ail servo lock when Rx is overloaded with movement. Other channels more stable.

I will set play delay to 0 for mid switch position.

closed 2021-05-10 by szempruw

szempruw, 2021-05-10:
I set the play delay to 0.

I ensured that the order of setup was done correctly. Bind, Calibrate Rx using Calibration Lua, Use setup Lua to choose wing type and mounting type (in my case Horizontal Reverse), and ensure SxR function and QM is enabled. My throttle was reversed so Thr forward was -100%. The 3-pos switch activated the Blue LED. You have to move sticks quickly or it will time out and go green and then must repeat.

At least for now, all the issues I had were related to not having at least the switch invoke the self check routine. Ensuring that the Thr is at or below -100% was key. Again, the self check had to be repeated a few times to ensure the joysticks were moved before the witch occurred. If not, I would only get, for example, partial elevator movement. 

Within RCDYI website written words, there was a comment made, that "depending on the software version used, the blue LED may not come on when Rx is powered up."

Saying all this, the Self Check was key as you suspected.

DaySpeed955i, 2021-05-10:
Glad it works now.
Once you have all the details in mind, setting up a SxR or SRx is pretty easy.

Enjoy.

szempruw, 2021-05-11:
Thank you. It works, except for the F/S button press which should initiate the Self Check. And no temporary blue led when powering up Rx. It's as if they got it going enough, but not yet complete to match the SxR manual on the FrSky web page. Oh well. We'll try a flight to see what happens. Thanks again.

DaySpeed955i, 2021-05-11:
In v2.x.x the F/S button is disable so is the blue light at power up :)