FrSkyRC/Firmware-Test #121: Telemetry not reliable for external S.Port sensors in Archer GR8

azaz44, 2021-05-24:
Telemetry is not reliable, when using FrSky Archer GR8 and external telemetry counters coming from the flight controller. This is not related to range, as this happens on the bench too.

Hardware:
- Horus X10s with ISRM update and recent firmware on it
- OpenTX 2.3.11
- FrSky Archer GR8 (different firmwares)
- Sprit 2 FBL (helicopter flight controller)

Similar setup, but with ACCST receivers, works fine.
However on Archer GR8 this happens:
- on firmwares 2.1.0 and 2.1.2:
  - telemetry works similar to ACCST
  - however, every few seconds data reception is interrupted and counter values are not updated
  - This happens every few seconds, and lasts for a couple of seconds
  - you can verify this by looking at the data
  - you can also verify this by checking "*" characters being shown on *Telemetry* screens
- on firmwares 2.1.5, 2.1.6, 2.1.7:
  - all counters are updated very rarely
  - update rate is so rate, that radio very often informs you "sensor lost"

Adelaxie, 2021-05-25:
[Archer-X_2.1.8-beta_20210521.zip](https://github.com/FrSkyRC/Firmware-Test/files/6535995/Archer-X_2.1.8-beta_20210521.zip)
Please confirm the fix. thanks.

azaz44, 2021-05-25:
@Adelaxie It's much better with this firmware:
- it behaves more like 2.1.0 and 2.1.2
- no "sensor lost"
- interruptions happen less often than in 2.1.0 and 2.1.2
- but they still do happen occasionally

Is this firmware save to fly with?

islandflight, 2021-05-25:
Does not fix the lag in telemetry data which seems to be affecting the 0Bxx telemetry addresses used by esc's

SmartTommy, 2021-05-26:
> Does not fix the lag in telemetry data which seems to be affecting the 0Bxx telemetry addresses used by esc's

which esc do you use? Neuron? could you show us a vedio? Thanks

SmartTommy, 2021-05-26:
> @Adelaxie It's much better with this firmware:
> 
> * it behaves more like 2.1.0 and 2.1.2
> * no "sensor lost"
> * interruptions happen less often than in 2.1.0 and 2.1.2
> * but they still do happen occasionally
> 
> Is this firmware safe to fly with?

could you show us a vedio about the "interruptions"? and what's the FC connect to the Rx

islandflight, 2021-05-26:
Hi telemetry lag has been tested using a neuron esc and RCLS telemetry converter. The lag is in the current, mah consumption and esc rpm.  There does not seem to be a lag when using a yge  esc which reports at telemetry address 0200

SmartTommy, 2021-05-26:
> Hi telemetry lag has been tested using a neuron esc and RCLS telemetry converter. The lag is in the current, mah consumption and esc rpm. There does not seem to be a lag when using a yge esc which reports at telemetry address 0200

what is the "RCLS"? we does not have this, why not connect the Neuron Sport to the GR8(firmware V2.1.8 ) sport?

islandflight, 2021-05-26:
I have tested with the neuron (original don't have s version) with v2.1.8 and delay there. This is a link to the RCLS telemetry converter. Great little device!  https://www.rclightsystems.com/product-page/esc-telemetry-converter

islandflight, 2021-05-26:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.youtube.com/watch%3Fv%3D62QExqsimi4&ved=2ahUKEwjljLnItObwAhWOHxQKHRSQAMsQwqsBegQIBBAB&usg=AOvVaw25dIMRmeHihOd_3PZJcoVR

islandflight, 2021-05-26:
Not a great quality video but you can see the esc current still falling some time after the throttle closed

SmartTommy, 2021-05-26:
> Not a great quality video but you can see the esc current still falling some time after the throttle closed

we find that, the ethos filter bring the lag, The next version will fix this. Thank you.

EngelMT, 2021-05-26:
@SmartTommy Only for your info. The guys reported that issue with X10S and OTX and not with ETHOS I think. So will be important to fix it in OTX too.

SmartTommy, 2021-05-26:
> @SmartTommy Only for your info. The guys reported that issue with X10S and OTX and not with ETHOS I think. So will be important to fix it in OTX too.

The filter problem answer only for @islandflight who use the X20.
we can't reproduce the problem @azaz44 has reported,  still waiting his vedio to show us, if you can reproduce that,please show us, it will help us to find the problem. Thanks!

azaz44, 2021-05-26:
@SmartTommy Here are videos.
----
Made with beta version `Archer-X_2.1.8-beta_20210521.zip` posted in this thread - telemetry counters. As you see there are times when they are not refreshed. Happens regularly, every few seconds, for few seconds.

https://youtu.be/HdIC4MaIELc

----
This one also made with beta `Archer-X_2.1.8-beta_20210521.zip` from this thread - but this is Spirit integration script. Radio reads data from the flight controller over telemetry anytime I switch the screen. As you see, it hangs sometimes. This does not happen on my X8R and X6R, both with ACCST. For me it's easiest way to see if there is a problem.

"Sensor lost" is normal in this case. When this script is running, radio does not send normal telemetry values, hence the message.

https://youtu.be/dkSSX9oH3TE

-----
This is a video from firmware version `2.1.0`

https://youtu.be/bYEsthtFpI0

-----
This is firmware version `2.1.7` (same happens with anything higher than 2.1.2)
Telemetry counters:
https://youtu.be/US0OnL_Bdss
They are refreshed so rarely that radio complains "sensor lost".

----
This is firmware version `2.1.7` (same happens with anything higher than 2.1.2)
Intergration script (completely not usable because of the problem)
https://youtu.be/5g5to5DQLsc

----
In general:
- The beta from this thread improves things greatly over 2.1.7.
- However, looking at the videos now, I'm not sure if it improves anything over 2.1.0?

SmartTommy, 2021-05-27:
> ## @SmartTommy Here are videos.
> Made with beta version `Archer-X_2.1.8-beta_20210521.zip` posted in this thread - telemetry counters. As you see there are times when they are not refreshed. Happens [...]

Thank you for your Vedios, in the first vedio(https://youtu.be/HdIC4MaIELc), i see some unknow telemetry id such as Vibe/Vpak...,what's the Sport Sensor do you used? RB10/20 or others?

In the second vedio (https://youtu.be/dkSSX9oH3TE), The lua script will preempt the telemetry transmission channel，so it will keep warning "Sensor Lost" untill the lua script finished.

azaz44, 2021-05-27:
> Thank you for your Vedios, in the first vedio(https://youtu.be/HdIC4MaIELc), i see some unknow telemetry id such as Vibe/Vpak...,what's the Sport Sensor do you used? RB10/20 or others?

Vibe and Vpak come from the flight controller, it is Spirit 2 FBL - http://spirit-system.com
Vibe was originally named "Fuel", but I renamed it, because it actually sends amount of model vibrations as a percentage. Comes from Spirit directly.
VPak is battery voltage, which I also renamed, probably original name was sth like RB1V? Not sure.. Spirit gets this data from the ESC (Scorpion Tribunus II 14-200 in my case).

OpenTx allows to rename the counters just for display.

> In the second vedio (https://youtu.be/dkSSX9oH3TE), The lua script will preempt the telemetry transmission channel，so it will keep warning "Sensor Lost" untill the lua script finished.

"Sensor lost" is fine in this case. Happens on ACCST too. The problem is with communication interruptions. You can see on some screens that the black progress bar at the top freezes and no new data is coming for some seconds. This does not happen on ACCST. This seems to be related to these "pauses" which telemetry screen has too, when updating counters.

azaz44, 2021-05-27:
> "Sensor lost" is fine in this case. Happens on ACCST too. The problem is with communication interruptions. You can see on some screens that the black progress bar at the top freezes and no new data [...]

More info -> you can see these interruptions at screens `2/7`, `5/7`, `7/7` (this is printed at the top right corner of the screen).

azaz44, 2021-05-28:
More info - here's how it looks like on ACCST v2. 
Receiver is X6R, updated to recent ACCST v2. Same transmitter and OpenTX as before.

tele counters: https://youtu.be/IktZgm254wM
integration script: https://youtu.be/ahHpDZpqgs8

SmartTommy, 2021-05-31:
> More info - here's how it looks like on ACCST v2.
> Receiver is X6R, updated to recent ACCST v2. Same transmitter and OpenTX as before.
> 
> tele counters: https://youtu.be/IktZgm254wM
> integration script: https://youtu.be/ahHpDZpqgs8

i checked the talk between the GR8 and Spirit pro, and find that the Spirit does't comply with the sport protocol, it should respond to  the Rx when the Rx send the Spirit PhyID,but the spirt does not reply（it respond every 100ms). it works well with the X6R, but the GR8(all of Archer) is faster, if the Spirit does not respond within 96ms, the Spirt PhyID will remove from work queue to the idle queue , this will lead the Rx send the Spirit PhyID every 384ms, that is the reseaon of the "interruptions ".

We will contact Spirit company。Thank you for your test!

Adelaxie, 2021-07-22:
Spirit System has fixed the issue on F.port.

closed 2021-07-22 by Adelaxie

azaz44, 2021-08-09:
@Adelaxie

I thing the issue might have been closed too early.

I tested this in latest Archer firmware (2.1.8) and latest Spirit (3.3.1) and the problem is still there. Telemetry now works, but has interruptions. So it's much better than 2.1.7, but same like just like 2.1.7_beta posted here, or bersions 2.1.2 and 2.1.0.

See video:

https://youtu.be/P7eM71Q1ja0

Is this still something to fix in Spirit, or in Archer firmwares?

Everything works without problems with two other receivers which I have, which are ACCST (X6R and X8R).

Adelaxie, 2021-08-12:
Recently, we are constantly receiving user feedback about abnormal telemetry errors caused by connecting a 3rd party or DIY sensor.

S.Port/F.Port protocols were authorised only to the companies who signed with us under an agreement which asks for technically confidential and samples provided to get compatible tests.
I will contact Spirit team again to suggest them to provide samples to us to check if the sensors adapts to the protocols well.

azaz44, 2021-08-14:
> Recently, we are constantly receiving user feedback about abnormal telemetry errors caused by connecting a 3rd party or DIY sensor.
> 
> S.Port/F.Port protocols were authorised only to the companies who signed with us under an agreement which asks for technically confidential and samples provided to get compatible tests.
> I will contact Spirit team again to suggest them to provide samples to us to check if the sensors adapt to the protocols well.

Then why not make backward compatible? I mean, they all work on ACCST.