FrSkyRC/Firmware-Test #123: Wrong speed reported by ASS-70 after flashing to latest firmware 20210318

ululi1970, 2021-05-28:
After updating the FW on ASS-70 speed sensor (FW version  20210318), the reported speed 
is wrong (too low). Previously, by blowing on the tube it would read 148 kts (270 km/h, the maximum for this sensor). Now it only reads 64 kts. Likewise, speed reported during flight is too low. 

Tested with both ACCESS and ACSST receivers. Same behavior. Can you please post the older firmware, so that I can roll back?

DaySpeed955i, 2021-05-30:
I am reporting the same issue, the firmware 20210318 give about half the real speed given by GPS. I have use the reset button on the ASS-70 before each run with same results.
I also reloaded the firmware twice, same results.

Tested with ACCESS and ACCST Rx, latest firmware, same results.

thanks.

ululi1970, 2021-06-21:
Come on FrSky, it is not hard to post an older firmware on your website....

mshagg, 2021-08-10:
Confirm this issue 100%

Have frsky abandoned users of the pitot sensors? It's completely broken for ACCESS receivers on the old firmware due to telemetry loss warnings and now useless on the March 2021 firmware due to grossly inaccurate readings.

Whats worse is that we can't even flash back to use the sensors with ACCST receivers.

Has anyone confirmed if the same issue effects the ass-100?

FBarrao, 2021-08-15:
This problem is not related with access by isself, my ASS100 works fine under access with X12 under opentx, but indicates about half speed with the X20S ( ethos)  tested this afternoon. 
https://www.rcgroups.com/forums/showpost.php?p=47638905&postcount=10361

Adelaxie, 2021-08-25:
Fixed on latest firmware updates.

closed 2021-09-07 by Adelaxie