FrSkyRC/Firmware-Test #124: VFR Values - Recommended Minimum Values for Telemetry Alerts

pld74, 2021-05-31:
Hello everyone, not sure where to post this so hoping this is the correct place, if not apologies and please direct me.

What is an acceptable VFR (Valid Frame Rate) minimum so that I can set telemetry alerts? Thank you!

### **Hardware**:
**Transmitters** 
Horus X12S, OpenTX v2.3.11 w/ISRM Upgrade Module v2.1.6
Tandem X20 Ethos v1.0.7, ISRM v2.1.8
**Receivers ** 
RX8R Pro v2.1.1
XM+ v2.1.2
R8 Pro v2.1.7
RS v2.1.8

pau33, 2021-08-07:
I use 50% with X9DSE+ and Archer receivers, but I also are interested in meaningful values.
Perhaps Frsky could give some explanations too.

pld74, 2021-08-12:
@frsky Any suggestions for an ideal VFR value for telemetry purposes? thanks

closed 2021-08-23 by pld74